system76/firmware-open #385: lemp11 won't charge battery and system76 ignored my support ticket

thesle3p, 2022-12-13:
- Model: lemp11 
- BIOS version: 2022-11-21_b337ac6
- EC version: 2022-11-21_b337ac6
- OS: pop!_OST 22.04 LTS
- Kernel:  Linux pop-os 6.0.6-76060006-generic #202210290932~1669062050~22.04~d94609a SMP PREEMPT_DYNAMIC Mon N x86_64 x86_64 x86_64 GNU/Linux

My brand new lemp11 won't charge the battery when plugged in to the AC adapter or a USB-C dock

### Steps to reproduce

I tried updating firmware it did not fix the problem
I tried plugging the laptop into two different USB-C docks it did not fix the problem
I tried plugging the laptop into the AC adapter it did not fix the problem
I tried cutting a support ticket and no one responded to the problem

### Expected behavior

the laptop should charge and system76 should respond to support tickets

### Actual behavior

the laptop does not charge and system76 has not responded to my support ticket

### Additional info
```
cat /sys/class/power_supply/BAT0/status 
Not charging
```
```
upower -d
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Tue 13 Dec 2022 08:19:37 AM EST (3827 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               Notebook
  model:                BAT
  serial:               0001
  power supply:         yes
  updated:              Tue 13 Dec 2022 09:21:38 AM EST (106 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              72.303 Wh
    energy-empty:        0 Wh
    energy-full:         73.92 Wh
    energy-full-design:  73.92 Wh
    energy-rate:         0 W
    voltage:             8.628 V
    charge-cycles:       1
    percentage:          97%
    capacity:            100%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Tue 13 Dec 2022 08:49:38 AM EST (2026 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              72.303 Wh
    energy-full:         73.92 Wh
    energy-rate:         0 W
    charge-cycles:       N/A
    percentage:          97%
    icon-name:          'battery-full-charging-symbolic'

Daemon:
  daemon-version:  0.99.17
  on-battery:      no
  lid-is-closed:   yes
  lid-is-present:  yes
  critical-action: HybridSleep
```

thesle3p, 2022-12-13:
106703 is the ticketID

ahoneybun, 2022-12-13:
Thank you we'll use that ticket for the support.

closed 2022-12-13 by leviport