system76/firmware-open #387: Oryp8 fails to boot properly when HDMI is pluggged in and cannot resume from suspend

n-hebert, 2023-01-19:
- Model: oryp8 
- BIOS version: 2022-10-14_4136ef8 
- EC version: match 
- OS: Pop!_OS 22.04 LTS 
- Kernel: 6.0.12-76060006-generic 

My Oryx Pro has a terrible time with power states unless HDMI is unplugged  (and maybe even USB) and it boots from power off. 
Suspending and resuming is basically fully broken. If I don't act very carefully, it will either brick (power off and cannot turn on for quite a few hold presses of power) or pretend to turn into a jet engine (black screen, keyboard lit, fan on maximum and loud -- no activity or response).

### Steps to reproduce

1. Power on the Oryx Pro with HDMI plugged in

or 

1. Unplug power cord from closed Oryx Pro and let suspend
2. Open to resume

### Expected behavior

Laptop powers on normally

### Actual behavior

A coin flip of a brick or a jet engine impression

### Additional info

It seems to happen when configured to use NVIDIA. If I only use Intel video then, of course, the HDMI isn't used so the first bug is muted. The second bug also seems ok? It seems to be a video hand-off issue.

n-hebert, 2023-01-28:
Happens on compute graphics, too. Updated the description.

n-hebert, 2023-02-07:
I've noticed some other issues such as having HDMI attached results in the operating system not getting the attached LCD as a display and suspend failing without having to change the power supply mode. I would normally open more  issues, but I believe the Oct '22 update was where things started breaking, so they may all be symptoms of this issue here.

Is it possible to flash the previous Oryp8 firmware to validate this? I don't see any archives where they would be stored. Could you help me try out the old firmware, e.g. @crawfxrd ?

I'm gung-ho to get troubleshooting this, hands-on, as it renders this laptop almost useless to me to have it so faulty. :disappointed:

crawfxrd, 2023-02-07:
- [oryp8-76ec_2021-09-23_b9b0e89.zip](https://github.com/system76/firmware-open/files/10678001/oryp8-76ec_2021-09-23_b9b0e89.zip)

You will need to reset the CMOS after flashing.

n-hebert, 2023-02-07:
Thanks kindly! Do you have the relevant steps for that for this model handy? I haven't done so yet :sweat_smile:

crawfxrd, 2023-02-07:
To use:

- Extract the ZIP to a FAT32 volume on a GPT partitioned USB drive
- Plug in the USB drive
- Boot the machine

n-hebert, 2023-02-07:
It's installed now. I can't seem to boot into the firmware interface or my second disk, though. Do I have to now do something to clear the CMOS as you had said?

```
$ sudo efibootmgr
Timeout: 2 seconds
No BootOrder is set; firmware will attempt recovery
$ sudo efibootmgr -c
Could not prepare Boot variable: No such file or directory
```

crawfxrd, 2023-02-07:
Yes. Unplug and replug the RTC battery.

n-hebert, 2023-02-07:
Thanks again @crawfxrd , did that and that sorted out that new oddity.

Seems the suspend bug is still here, live and kicking. It's such an unpredictable bird that I won't stare too much at cosmetic differences.

Interestingly, the other issue I was suggesting may be conjoint with it *is not*. The HDMI-on-boot causes a different issue.
I'm trying to tease apart what the issues should be here, but I know that booting with anything plugged in on the latest firmware gives a whole host of issues, as described, it just seems that there's a second defect to do with video. I'll open that one now.

crawfxrd, 2023-08-09:
2023-07-19_05cfb1a has been released.

I was not able to reproduce this issue.

n-hebert, 2023-08-09:
Hey, me neither.  Looks like a job well done, at first testing, @crawfxrd :partying_face: 

I tried several suspend-resumes on Pop!_OS 22.04 LTS on Hybrid graphics and it comes back up whether or not I have the plug in or out or switch it from one to the other.
Thanks for the fix!

If I end up spotting any lingering behaviour in the other display modes, I can comment or open a new issue :+1: 

Thanks again :tada:

closed 2023-08-09 by n-hebert