system76/firmware-open #391: Oryp8 latest firmware fails to properly provide internal LCD to the host OS and only the external can be used

n-hebert, 2023-02-07:
- Model: oryp8 
- BIOS version: affected version is 2022-10-14_4136ef8 
- EC version: I flashed [oryp8-76ec_2021-09-23_b9b0e89.zip](https://github.com/system76/firmware-open/files/10678001/oryp8-76ec_2021-09-23_b9b0e89.zip) manually on #387 -- not sure I'm filling this out right :smile: 
- OS: Pop!_OS 22.04 LTS  and Windows 10 
- Kernel: 6.0.12-76060006-generic 

On the latest firmware for the Oryp8 (from Oct 22), if you boot with HDMI plugged in a whole host of video related problems erupts on Pop!_OS that are hard to describe, but effectively summarize as the title given. On Windows the problem is much simpler to describe: the LCD panel is not detected at all. If you press Windows+P or whatever hot key you like to bring up the options to use a 2nd screen, it says it's using the "PC Screen" when it's using the external and selecting any other options does nothing. If you unplug the external, the LCD screen does not come back on; the only screen that is addressable is the external.

To fix, a reboot is required.

As part of troubleshooting #387, I flashed back to the previous firmware from Sept '21 and this problem *does not* exist there. It's functional and better.

### Steps to reproduce

1. Flash the latest Oct '22 firmware to the Oryp8
2. Plug in a live HDMI display to the Oryp8 (the screen must be on and ready due to communication between the devices)
3. Power on (preferably Windows) and let the system boot

### Expected behavior

The LCD panel can be used along with the external monitor

### Actual behavior

On Windows, the internal LCD panel is not seen, even in device manager.
On Pop!_OS the LCD panel just shows the boot logo forever or goes black or worse things happen between the both of the screens.

### Additional info

This has persisted across multiple updates & upgrades of both Pop!_OS and Windows 10 and I do not have a work-around other than a hard reboot.

crawfxrd, 2023-08-09:
2023-07-19_05cfb1a has been released.

Need to check if this still occurs.

n-hebert, 2023-08-09:
Good news!

I'll lend a hand and try both this and the other, alongside ya.

n-hebert, 2023-08-09:
Looks like this one is still a live fish for me.

Boot with HDMI, load into Windows 10 (default selection), now you cannot use the LCD no matter what settings you change.
Unplugging the HDMI just leaves a black screen.
The screen is *lit* if that's interesting. i.e. there's voltage going to it. It's just showing nothing and no device picks up.
This isn't a Windows-only issue; it occurs on NixOS for me, too. Obviously muted when integrated or compute graphics are selected, as there's no HDMI. 

In short, identical behavior. :confused:

n-hebert, 2024-03-09:
2023-09-08_42bf7a6 still exhibits identical behavior.
I would really like to see a fix on this one. I have reason to believe this is a regression introduced in the latest 2021 firmware -- the laptop didn't do this when I bought it initially. Maybe the old bugs that it replaces are worse, but I hope that helps to triage.

crawfxrd, 2024-03-09:
Please provide the cbmem log:

```
make -C coreboot/util/cbmem/
sudo ./coreboot/util/cbmem/cbmem -c > cbmem.txt
```

n-hebert, 2024-03-10:
[cbmem.txt](https://github.com/system76/firmware-open/files/14548338/cbmem.txt)

Happy to. Hope it helps, let me know if I need to boot into the fault mode before running that, this is just from my system running without first reproducing the defect. Call it a 'good boot' log.

crawfxrd, 2024-03-10:
Yes. Cold boot into a state that produces the issue and get the cbmem log.

n-hebert, 2024-03-11:
Hey @crawfxrd , done. Here's the bad boot for CB Mem.
On Pop!_OS the display doesn't even come up, despite waiting a long time.
It looks like this:
[badboot cbmem.txt](https://github.com/system76/firmware-open/files/14564891/badboot.cbmem.txt)
[image]

I hope it makes clear the degree of the bug: if you plug in HDMI before boot then you just can't use the laptop.
`Linux 6.6.10-76060610-generic #202401051437~1709085277~22.04~31d73d8 SMP PREEMPT_DYNAMIC Wed F x86_64 x86_64 x86_64 GNU/Linux` on Pop!_OS LTS (22.04)