system76/firmware-open #393: Open Firmware renders Lemp9 unbootable after commit b337ac6.

XV-02, 2023-02-21:
While testing https://github.com/system76/ec/pull/338, I managed to render a Lemp9 unbootable after flashing an up-to-date [85f8a8b](https://github.com/system76/firmware-open/commit/85f8a8b933ea531bdd5677d226f68b77de5c3579) firmware-open dirty build with the PECI EC branch checked out.

Attempting to flash a clean 85f8a8b build did not recover the system, and I eventually recovered it by flashing a build from commit [b337ac6](https://github.com/system76/firmware-open/commit/b337ac66fb3b8d77880e81f35acf3c50072a85c2).

Subsequently, I tested the 85f8a8b-dirty build again to the same results, and recovered it the same way. Testing a clean 85f8a8b build also rendered the system unusable, as did a clean [6562cf2](https://github.com/system76/firmware-open/commit/6562cf213d0dd0602fb414a574e6a37962590be0) build. 

After flashing any of the builds rendering the system unusable, I noticed the following behaviour:
1. After power-on, the system status light was solid green - consistent with being in a powered on state - but the display was blank and remained blank, and there was no evidence of response to input. This state was persistent even when left for several minutes.
2. After powering the system down by holding the power button, the CPU cooler was hot to the touch, despite the fans never kicking in.
3. Both USB and CH341A flashing caused the problem.
4. Recover required using a CH341A programmer.

XV-02, 2023-02-28:
Having tested firmware on Galp4, I'm seeing the same behaviour as I did on Lemp9. Builds after the 4.19 rebase seem unable to boot on 10th gen Intel. I did note that I was able to establish a state where a post 4.19 rebase EC version was reported with the bios reporting as b337ac6 on a functional boot, though that was more the result of the recovery process. 

I again had to build from b337ac6 and flash with the CH341A programmer.

nvandal, 2023-03-28:
Also same issue when testing 799ed79 from [new-models](system76/firmware-open/tree/new-models) on lemp9 trying to check out secure boot. Recovered with CH341A flashing b337ac6.

XV-02, 2023-07-27:
At current, this is a complete list of 10th gen and older systems for which firmware can be built from System76 repositories:

- [ ] Adder 1 (addw1)
- [ ] Adder 2 (addw2)
- [ ] Bonobo 14 (bonw14)
- [ ] Darter Pro 5 (darp5)
- [ ] Darter Pro 6 (darp6)
- [ ] Galago Pro 2 (galp2)
- [ ] Galago Pro 3 (galp3)
- [ ] Galago Pro 3 (galp3-b)
- [ ] Galago Pro 3 (galp3-c)
- [ ] Galago Pro 4 (galp4)
- [ ] Gazelle 14 w/ Nvidia 1650 (gaze14_1650)
- [ ] Gazelle 14 w/ Nvidia 1660 Ti (gaze14_1660ti)
- [ ] Gazelle 15 (gaze15)
- [ ] Lemur Pro 9 (lemp9)
- [ ] Oryx Pro 5 (oryp5)
- [ ] Oryx Pro 6 (oryp6)
- [x] Oryx Pro 7 (oryp7)

Entries with check marks are (as of 2023-07-27) known to work with at least one post https://github.com/system76/firmware-open/commit/b337ac66fb3b8d77880e81f35acf3c50072a85c2 build from this repo. 

So, for the moment, I would recommend running `sudo dmidecode -t1` and seeing if the laptop version is checked off before attempting to flash a post https://github.com/system76/firmware-open/commit/b337ac66fb3b8d77880e81f35acf3c50072a85c2 build.

crawfxrd, 2023-07-27:
coreboot-based firmware was never released for the following and so are not blockers:

- KBL (galp2, galp3, galp3-c)
  -  galp3-c has a firmware PR
- WHL (darp5, galp3-c)
  - both have firmware PRs
- CFL (addw1, gaze14, oryp5)
  - oryp5 has a firmware PR

crawfxrd, 2023-08-01:
lemp9:

```
[INFO ]  POST: 0x92
[INFO ]  POST: 0x98
[INFO ]  POST: 0xe3
[EMERG]  FspMemoryInit returned with error 0x80000007!
```

crawfxrd, 2023-08-01:
https://github.com/system76/coreboot/pull/190 will fix booting on CML-U and unblock making new firmware releases for them.

PR #444 (2023-08-01): Update EC and coreboot
Brings in the changes to resolve:

- Resolves: #78
- Resolves: #393
- Resolves: #424
closed 2023-08-01 by jackpot51