system76/firmware-open #395: Lemp11 (and probably others). Add option to disable Intel VT and VT-d in bios settings

JordanBoulan, 2023-03-08:
I just bought a brand new lemp 11. I was disappointed by the lack of settings in the bios. I know you can turn off virtualization in the linux boot cmd line but you can't do that for Windows 10.

I was planning on dual-booting linux and Win10 on separate drives. You can turn off hyper-v but with virtualization still enabled any software malicious or otherwise can take advantage of and enable virtualization at run-time if running in the kernel to subvert the running OS/Windows and become the hypervisor.

It seems like a simple option to add disabling it in the BIOS. This would help with the security of running Windows a lot if you are not using virtualization. As far as I know you can't fully disable it at boot time on Windows only Hyper-V.

Please consider adding this option in the bios. **There needs to be an option to disable VT and VT-D in the bios so Windows installations can run with virtualization fully off. Right now there is no way to disable virtualization fully while running windows 10, which is a issue.**

Thank you!

JordanBoulan, 2023-03-09:
Would this be possible to add or is there a hardware limitation that prevents it. Also if there is a way to disable virt on Windows fully I dont know of please let me know.

Thanks

crawfxrd, 2023-03-09:
It's hard-coded to be enabled in coreboot (`VtdDisable` in `src/soc/intel/alderlake/romstage/fsp_params.c`).

JordanBoulan, 2023-03-10:
Does the auto-flasher (system76-firmware --open tool or whatever its called support custom builds? I also don't want to have to re-flash between OS's. It doesn't seem that hard to add? Is there any reasoning behind it being hard-coded. Virtualization is a major security issue if enabled or unused or used improperly.

JordanBoulan, 2023-03-17:
@crawfxrd I guess to follow up? Is there any reason this can't be added to the bios menu? I could contribute and add menu option and code to the UI. Can you point me towards the UEFI ui code? I could maybe find it if I looked. I first wanted to know if there is any reason this can't be a menu option and should I contribute and make it one?

To flash custom builds do you need like an SPI flasher and have to open up the laptop or is there an easier way to do it?

crawfxrd, 2023-03-18:
coreboot is primarily statically configured. There first needs to be a mechanism to be able to modify the FSP params that isn't CMOS options; There's an MR for [using edk2 variables for option values](https://review.coreboot.org/c/coreboot/+/62562). Then an edk2 module/library/driver (probably all 3) is needed for modifying the FSP values.

For flashing instructions, see [`flashing.md`](https://github.com/system76/firmware-open/blob/master/docs/flashing.md).