system76/firmware-open #396: Trusted Execution Technology support

DemiMarie, 2023-03-19:
This can be used for Dynamic Root of Trust for Measurement and will enable Qubes OS Anti-Evil Maid (AEM) support in the future.

crawfxrd, 2023-06-28:
Going beyond this, we probably to to ensure compatibility with Intel Converged Boot Guard and Trusted Execution Technology (CBnT).

9elements has implemented support for both in coreboot.

- `INTEL_TXT`
- `INTEL_CNBT_SUPPORT` (selects `INTEL_TXT`)

tlaurion, 2024-04-01:
From https://doc.coreboot.org/security/vboot/measured_boot.html#known-limitations

> At the moment measuring IBB dynamically and FMAP partitions are not possible but will be added later to the implementation.

> Also SoCs making use of VBOOT_RETURN_FROM_VERSTAGE are not able to use the measured boot extension because of platform constraints.

While https://doc.coreboot.org/security/intel/txt_ibb.html

Seems pretty complete, just unsure if coreboot permits to configure IBB from kconfig options directly. 

What is currently missing from coreboot side? Which platforms and CPU family are the lowest requirements to implement TXT with sinit+acm to measure bootblock as part of IBB with CPU anchored RoT @DemiMarie?

DemiMarie, 2024-04-01:
@tlaurion I’ll leave that question to 3mdeb engineers.

tlaurion, 2024-04-01:
> @tlaurion I’ll leave that question to 3mdeb engineers.

Posted 
- https://matrix.to/#/!rsKWMJGPMsyPTTjXuh:matrix.org/$ulC6_QitP0vaRQfKmbS1475QnhggPtBEjlqg_kJwy3M?via=matrix.org&via=nitro.chat&via=invisiblethingslab.com
- https://matrix.to/#/!UhZSUDKFQINYdOEfGo:matrix.org/$nbwuu_H2SCLjApKRQp2VbCgk2k8JGFmQXwwvkQdIPSg?via=matrix.org&via=nitro.chat&via=hackliberty.org