system76/firmware-open #399: Darter Pro (darp8) Keyboard backlight turns on after suspend

DrymarchonShaun, 2023-03-21:
- Model: darp8
- BIOS version: 2022-11-21_b337ac6
- EC version: 2022-11-21_b337ac6
- OS: Arch Linux
- Kernel: 6.2.7-arch1-1

Right after suspending (closing lid, fn + f12, `systemctl suspend`) the screen and keyboard backlight turn off, but after a moment the keyboard backlight turns back on.

### Steps to reproduce

Step 1 - Set backlight brightness to on `sh -c 'echo "255" | sudo tee /sys/class/leds/system76_acpi::kbd_backlight/brigh
tness'`
Step 2 - Suspend `systemctl suspend`
Step 3 - Backlight will turn off for a few seconds, then it will turn back on
Step 4 - Resume, backlight will turn back off.

### Expected behavior

On suspend, keyboard backlight should turn of and stay off.

### Actual behavior

On suspend, keyboard backlight turns off for a few second, then turns back on, while the device stays suspended, (green light is flashing.)

### Additional info

This has already been reported on Pop!_OS as well - [https://www.reddit.com/r/System76/comments/11hl1sx/darterpro_keyboard_backlight_turns_on_after/](https://www.reddit.com/r/System76/comments/11hl1sx/darterpro_keyboard_backlight_turns_on_after/)

leviport, 2023-03-21:
Did this just suddenly start? I'm unable to replicate this behavior on my lab unit.

DrymarchonShaun, 2023-03-21:
I only just noticed it within the past days, I'm not sure if it started recently or not.

leviport, 2023-03-21:
It might be good to open a support ticket for this issue: https://support.system76.com/

How long have you had the machine?

leviport, 2023-03-21:
@uSpike I'm pretty sure that's a different issue. In that one, the color reverts to white when resuming from suspend. In this one, the keyboard backlight is turning on while the machine is suspended. Also, that other bug is not specific to darp8.

uSpike, 2023-03-21:
Ah I see, well then I misspoke in https://github.com/system76/ec/issues/343#issuecomment-1473841647 - what I see is the issue documented here.  I see the backlight turn on when suspended, and off when woken.  It is not white but instead the color which was last set.

leviport, 2023-03-21:
@uSpike did this start happening all of a sudden, and if so, when?

DrymarchonShaun, 2023-03-21:
> How long have you had the machine?

got it sometime around October 2022

leviport, 2023-03-22:
I'm now seeing this happen on my lab unit. I suspect it's related to the newer kernel. Now that I can reproduce the bug, we'll work on a fix.

crawfxrd, 2023-03-29:
Does it still happen on 6.2.8?

leviport, 2023-08-23:
Since Pop now is on 6.4.6, I'm assuming that this can be closed.

rkintzi, 2023-09-10:
I still have the same issue on the linux 6.4.12.arch1-1. The rest of the system (BIOS/EC) is the same as in the original thicket. Every time I suspend the system keyboard backlight turns initially off and after a few second it goes on.  It is restored to original level after resuming from suspend.

DrymarchonShaun, 2023-09-10:
@rkintzi I'm currently on 6.5.1 with no issues. Is there any particular reason you're still on 6.4.12?

rkintzi, 2023-09-10:
6.5.2 doesn't solve the problem for me. I don't have 6.5.1 at hand, but will try to build later.

My experience is that darp8 is very unstable (or maybe I should say that Linux kernel is very unstable on darp8), so I tend to keep a working kernel release. I didn't even notice the problem until yesterday's evening.

DrymarchonShaun, 2023-09-11:
I'm on NixOS now, so I can't properly confirm if It's started happening again on Arch. If 6.5.2 isn't working, there's probably no point building 6.5.1.

DrymarchonShaun, 2023-09-15:
@rkintzi I take back what I said, I just noticed my laptop has started doing this again.

@leviport probably need to reopen this.

May be unrelated, but I did update to kernel version 6.5.2

leviport, 2023-09-15:
>May be unrelated, but I did update to kernel version 6.5.2.

I think the last time this happened, it was resolved by a Linux update. If I'm correct about that, then that's probably not unrelated at all.

DrymarchonShaun, 2023-09-15:
I switched back to 6.5.1 and it's still occurring. I'm not sure what happened, maybe it has been doing it and I just hadn't noticed.

DrymarchonShaun, 2023-09-30:
So I'm honestly not sure what's happening anymore. Somewhere between [NixOS/nixpkgs@e5f018cf150e29aac26c61dac0790ea023c46b24](https://github.com/NixOS/nixpkgs/tree/e5f018cf150e29aac26c61dac0790ea023c46b24) and [NixOS/nixpkgs@5d017a8822e0907fb96f7700a319f9fe2434de02](https://github.com/NixOS/nixpkgs/tree/5d017a8822e0907fb96f7700a319f9fe2434de02)
My darp8 started to seemingly not fully suspend, the green light that should be flashing is staying solid.
For a good while when i was trying to figure out what that was about, I thought this issue may have been related as at one point when using the older e5f018cf150e29aac26c61dac0790ea023c46b24 revision, the backlight actually did stay off while properly entering sleep, (the green light started flashing.) But then with no changes, later on I tried again, and the backlight turned off and then turned back on, while the green light started flashing. So this original issue is seemingly being inconsistent, and now there seems to be a regression in *something* that's stopping my system from properly sleeping.

DrymarchonShaun, 2023-10-02:
Potentially of note: I noticed my system does seem to be reaching C10 
```
root@natrix ~# cat /sys/kernel/debug/pmc_core/package_cstate_show -pp                         
Package C2 : 16372577488
Package C3 : 10295
Package C6 : 66017
Package C7 : 0
Package C8 : 29237
Package C9 : 0
Package C10 : 98057701786
```

I don't fully understand s0ix and c-states, but from what I gather, isn't C10 a state beyond s0ix? Does that mean it is actually going to sleep properly, and just not being picked up correctly by whatever controls the backlight and power LED?

rkintzi, 2023-10-02:
The problem with s0ix and darp8 is know issue (I reported it back in 2022): #384

DrymarchonShaun, 2023-10-02:
> The problem with s0ix and darp8 is know issue (I reported it back in 2022): #384

I saw that, but for the longest time it wasn't an issue, and just recently started back up.