system76/firmware-open #400: `galp6` suspending with USB-C Thunderbolt dock does not work

ahoneybun, 2023-03-23:
Steps to reproduce the behavior:

I rebooted and selected the 6.x kernel, this allowed me to suspend as described earlier. I created 4 logs at the following points in time in a sequence:

Just after restarting and logging in, no suspend attempts
After successfully suspending and waking back up from the suspend
Plugging in the USB-C cable (no additional suspend)
Removing the USB-C cable, then attempting to suspend (unsuccessfully)
Expected behavior
System suspends and resumes correctly

This was on latest firmware 2022-11-21_b337ac6 and confirmed on the laptop unit.

comminutus, 2023-07-09:
Any idea when this might be prioritized?

curiousercreative, 2023-07-13:
galp5 similarly is unable to suspend when connected to [LG 40" UltraWide](https://www.lg.com/us/monitors/lg-40wp95c-w-ultrawide-monitor). Here's an excerpt from dmesg
```
$ sudo dmesg
[11797.362124] wlp0s20f3: deauthenticating from 2c:30:33:f1:3a:68 by local choice (Reason: 3=DEAUTH_LEAVING)
[11798.786017] PM: suspend entry (s2idle)
[11798.789889] Filesystems sync: 0.003 seconds
[11798.828636] Freezing user space processes
[11798.831432] Freezing user space processes completed (elapsed 0.002 seconds)
[11798.831435] OOM killer disabled.
[11798.831436] Freezing remaining freezable tasks
[11798.832324] Freezing remaining freezable tasks completed (elapsed 0.000 seconds)
[11798.832327] printk: Suspending console(s) (use no_console_suspend to debug)
[11799.088368] psmouse serio1: Failed to disable mouse on isa0060/serio1
[11799.640299] ACPI: EC: interrupt blocked
[11799.642768] ------------[ cut here ]------------
[11799.642769] thunderbolt 0000:00:0d.2: RX ring 0 already stopped
[11799.642832] WARNING: CPU: 1 PID: 38682 at drivers/thunderbolt/nhi.c:733 tb_ring_stop+0xa2/0x1f0 [thunderbolt]
[11799.642850] Modules linked in: hid_logitech_hidpp uhid tls snd_usb_audio snd_usbmidi_lib usbhid cdc_acm rfcomm snd_seq_dummy snd_hrtimer nf_conntrack_netlink xfrm_user xfrm_algo xt_addrtype br_netfilter vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) ccm cmac algif_hash algif_skcipher af_alg xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 wireguard curve25519_x86_64 libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel nf_tables nfnetlink bridge stp llc overlay bnep zstd zram zfs(POE) zunicode(POE) zzstd(OE) zlua(OE) zavl(POE) icp(POE) zcommon(POE) znvpair(POE) spl(OE) snd_hda_codec_hdmi snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match
[11799.642884]  snd_hda_codec_realtek snd_soc_acpi soundwire_bus snd_hda_codec_generic ledtrig_audio snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine joydev snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi intel_tcc_cooling snd_seq_midi_event iwlmvm snd_rawmidi uvcvideo videobuf2_vmalloc x86_pkg_temp_thermal videobuf2_memops intel_powerclamp mei_pxp mei_hdcp intel_rapl_msr videobuf2_v4l2 coretemp snd_seq nls_iso8859_1 btusb mac80211 btrtl kvm_intel videodev btbcm btintel snd_seq_device libarc4 bfq videobuf2_common btmtk processor_thermal_device_pci_legacy snd_timer kvm iwlwifi processor_thermal_device processor_thermal_rfim bluetooth input_leds mc cmdlinepart processor_thermal_mbox ecdh_generic rapl mei_me ecc snd spi_nor processor_thermal_rapl intel_rapl_common cfg80211 mei mtd hid_multitouch soundcore 8250_dw ee1004 intel_cstate int340x_thermal_zone serio_raw intel_soc_dts_iosf igen6_edac intel_pmc_mux typec mac_hid intel_hid
[11799.642919]  intel_scu_pltdrv sparse_keymap sch_fq_codel kyber_iosched msr parport_pc ppdev lp parport ramoops pstore_blk reed_solomon pstore_zone efi_pstore ip_tables x_tables autofs4 btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) i915 drm_buddy i2c_algo_bit ttm drm_display_helper cec rc_core drm_kms_helper crct10dif_pclmul syscopyarea crc32_pclmul rtsx_pci_sdmmc polyval_clmulni sysfillrect hid_generic polyval_generic sysimgblt ghash_clmulni_intel sha512_ssse3 aesni_intel crypto_simd nvme cryptd r8169 i2c_i801 spi_intel_pci psmouse rtsx_pci drm realtek thunderbolt nvme_core spi_intel i2c_smbus intel_lpss_pci intel_lpss xhci_pci idma64 nvme_common xhci_pci_renesas i2c_hid_acpi i2c_hid hid video wmi pinctrl_tigerlake system76_acpi(OE)
[11799.642954] CPU: 1 PID: 38682 Comm: kworker/u16:4 Tainted: P S         OE      6.2.6-76060206-generic #202303130630~1689015125~22.04~ab2190e
[11799.642956] Hardware name: System76 Galago Pro/Galago Pro, BIOS 2022-11-21_2021-07-20_93c2809-106-gb337ac6-dirty 07/11/2023
[11799.642957] Workqueue: events_unbound async_run_entry_fn
[11799.642962] RIP: 0010:tb_ring_stop+0xa2/0x1f0 [thunderbolt]
[11799.642972] Code: 0f 84 5c 01 00 00 48 81 c7 d0 00 00 00 e8 56 84 0a dd 45 89 e0 4c 89 e9 4c 89 f2 48 89 c6 48 c7 c7 7b 63 44 c0 e8 de fc 6d dc <0f> 0b 48 89 df 48 83 c3 50 e8 f0 e0 6b dd 48 8b 7b b8 e8 37 e1 6b
[11799.642974] RSP: 0018:ffffc1deaa293cf0 EFLAGS: 00010046
[11799.642975] RAX: 0000000000000000 RBX: ffff9f6941333300 RCX: 0000000000000000
[11799.642976] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[11799.642977] RBP: ffffc1deaa293d10 R08: 0000000000000000 R09: 0000000000000000
[11799.642977] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[11799.642978] R13: ffffffffc0446250 R14: ffff9f69419e6340 R15: 0000000000000000
[11799.642979] FS:  0000000000000000(0000) GS:ffff9f709e640000(0000) knlGS:0000000000000000
[11799.642980] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[11799.642981] CR2: 00000d2c0136d000 CR3: 0000000793210006 CR4: 0000000000770ee0
[11799.642982] PKRU: 55555554
[11799.642983] Call Trace:
[11799.642984]  <TASK>
[11799.642986]  tb_ctl_stop+0x3c/0xf0 [thunderbolt]
[11799.642997]  tb_domain_suspend_noirq+0x4e/0x70 [thunderbolt]
[... cut ...]
[11887.846513] wlp0s20f3: RX AssocResp from 2c:30:33:f1:3a:68 (capab=0x1011 status=0 aid=1)
[11887.853305] wlp0s20f3: associated
[11887.919251] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[11974.127847] input: MX Master 3 as /devices/virtual/misc/uhid/0005:046D:B023.0008/input/input34
[11974.128278] logitech-hidpp-device 0005:046D:B023.0008: input,hidraw5: BLUETOOTH HID v0.15 Keyboard [MX Master 3] on 18:cc:18:98:9c:c4
[11974.407652] logitech-hidpp-device 0005:046D:B023.0008: HID++ 4.5 device connected.
```

comminutus, 2023-12-06:
Checking in again, will this ever be fixed?

curiousercreative, 2023-12-06:
This bug is frustrating and I'm fortunate that I do not dock/undock often (I'm typically mobile as I have a desktop and only dock occasionally). I was previously docking/undocking more often and I created [a pair of scripts for docked vs undocked](https://gist.github.com/curiousercreative/e969f0e5ad5615b90cdcda61d67fd14f) and at the time I was changing quite a lot (scale factor, keymapping, etc) but now it's mostly just power modes and [toggling auto-suspend](https://gist.github.com/curiousercreative/e969f0e5ad5615b90cdcda61d67fd14f#file-dockify-sh-L6). Maybe there's something you can glean from it, I know others have leveraged UDEV to automate this as well.

comminutus, 2023-12-06:
@curiousercreative thanks for sharing. I did take a look through the scripts but I don't think it'll help in this case.  I believe the problem is fundamentally in the firmware.  I've got so far as to switch my entire dock to an HDMI-based one, and plug-in an HDMI cable, power cable, and USB cable, which is annoying instead of just plugging in one thunderbolt cable but at least I can use my workstation setup with the laptop and not lose my suspend/resume capability.

curiousercreative, 2023-12-06:
@comminutus yeah, I opted to just disable auto suspend and key Thunderbolt. Of course there is a firmware issue that needs fixing, but I'm on year 3 of living with it.