system76/firmware-open #405: "Cancel" at security code confirmation screen reboots back to security code confirmation screen

leviport, 2023-03-31:
- Model: serw13
- BIOS version: 2023-03-31_7254442
- EC version: 2023-03-31_7254442
- OS: Pop!\_OS 22.04
- Kernel: 6.2.6

Selecting "Cancel" at the screen where I enter an 8-digit number before updating firmware briefly shows the flashing screen, then reboots, but instead of taking me back to the OS, I'm taken back to the security code screen again.

### Steps to reproduce

While flashing internally, instead of entering a code and confirming, arrow down to the "Cancel" option and hit enter

### Expected behavior

I would expect "Cancel" to cancel the firmware update, restore the boot entries to normal, and take me back to a normal OS boot.

### Actual behavior

Confirmation screen doesn't go away until I enter the code and flash firmware

### Additional info

crawfxrd, 2023-05-30:
The problem is that when the unlock is canceled (which happens in firmware-setup), firmware-update is not removed from the boot override.

- Boot override is added to boot to firmware-update
- firmware-update sees firmware is locked, requests unlock
- System reboots to unlock the firmware
- User cancels unlock, system requests lock
- System reboots to lock the firmware
- System boots from override to firmware-update

crawfxrd, 2023-05-30:
Because it currently prompts to unlock before it prompts to install, hitting e.g., `Backspace` in firmware-update will cancel the update, but reboot to the firmware unlock screen (only once though instead of getting stuck).

closed 2023-06-28 by crawfxrd