system76/firmware-open #406: Lemp 11 - ESC sometimes just doesn't work to get into UEFI BIOS and/or after in UEFI settings keyboard/input/arrow keys doesn't work

JordanBoulan, 2023-03-31:
Basically, just the title.

Issue 1: I've noticed that sometimes hitting ESC just simply does nothing, even when mashing. This issue often seems to appear right before issue 2. Or in other words issue two seems to immediately follow issue 1 after rebooting 2-3 times to get issue 1 fixed.

Issue 2: There is also this bug where it goes into the settings but then the keyboard and arrow keys don't work and you can't do anything and are forced to reboot. This no input working in UEFI seems to persist across multiple reboots and I find myself having to reboot the laptop 5+ times just to get into UEFI with input actually working so I am able to select a boot option.

Anyone else have this issue?

leviport, 2023-03-31:
When this happens, does a USB keyboard work normally? This would help determine whether it's just a problem with the built-in keyboard, which could be an EC thing, or something else entirely.

dni, 2023-04-29:
seeing the same issue the first time after updating firmware on pang12, very frustrating, for me i yet wasnt able to get into bios

dni, 2023-04-29:
not sure what happened, but turning off and on the ps2 support inside BIOS fixed this issue for me. Also the issue i had why i updated firmware in the first place, was because the touchpad suddenly stopped working, which also gone away.

leviport, 2023-05-01:
@dni the pang12 does not have open firmware, so this issue is unrelated to any of that.

crawfxrd, 2023-07-24:
May be solved by https://github.com/system76/ec/pull/377.

closed 2023-09-19 by crawfxrd

dni, 2023-09-20:
i found my issue is probably hardware related one. everytime it happens i apply some pressure on the top right corner of the touchpad and it starts to work again for some time. pretty annoying, but it works.

leviport, 2023-09-20:
Sounds like re-seating the FFC connections to the touchpad might help. Fonzi-style workarounds tend to stop working after a while.