system76/firmware-open #407: `oryp9/10` HDMI

thomas-zimmerman, 2023-04-03:


brs17, 2023-04-03:
Do you have an idea when this issue was first and most recently reported?

VWY-github, 2023-04-05:
I have this problem about 40% of the time. Seems to be hit or miss. Much more likely to get to the OS if *not* connected to external monitor via hdmi (but sometimes it does work for me).
Hoping for a new bios version soonn (wasnt  a new version announced just today?) that might cure some ills.
Also have frequent issues with crypto, but that may be a different story

VWY-github, 2023-04-14:
When will new firmware be pushed out and automatically installable?

ahoneybun, 2023-04-24:
@VWY-github once it has passed QA, you will get a notification once it is available.

n-hebert, 2023-04-24:
Is anyone assigned to this one, @ahoneybun ? GitHub doesn't have anyone listed right now.

I reported a very similar this one (one may be a duplicate) via #391 and am very much looking to assist with squashing it in any way possible (r&d, impl, debug, whatever's appropriate). 
I'm very new to this area of the stack, but a veteran of software, is it fair to say this problem is in the EC rom? That's where I suspect it to be and was planning on digging through old commits, as we know this broke between 2021 and now.

I would love to collaborate with whoever is working on this!

leviport, 2023-06-27:
Oryp9 and oryp10 updates have been released.

closed 2023-06-27 by leviport

n-hebert, 2023-06-27:
hey @leviport , thanks for updating those. The Oryp8 also has the same bug, so a couple questions there:
1. Any update coming for the Oryp8?
2.  should we open a new issue for it or is #391 good enough to represent this?
Cheers

leviport, 2023-06-27:
#391 should be fine for tracking that. We'll have updates for oryp8 as soon as we can.

marcin-rzeznicki, 2023-06-28:
And how is that actually solved? For me it's still hit-miss to boot with HDMI plugged in