system76/firmware-open #414: addw3 fails to build due to missing coreboot/3rdparty/fsp contents

bcran, 2023-05-20:
- Model: addw3
- BIOS version: 2023-04-28_a220704
- EC version: 2023-04-28_a220704
- OS: Ubuntu 22.04.2 LTS
- Kernel: 6.2.6-76060206-generic

Running ./scripts/build.sh addw3 fails because .git/modules/coreboot/config has "update=none" for 3rdparty/fsp (I haven't checked if the build will subsequently fail due to other missing submodule contents).
The error is:
```
3rdparty/intel-microcode/intel-ucode/06-97-02 3rdparty/intel-microcode/intel-ucode/06-97-05 3rdparty/intel-microcode/intel-ucode/06-b7-01
    CC         postcar/arch/x86/exit_car.o
    CC         postcar/arch/x86/gdt_init.o
    CC         postcar/arch/x86/idt.o
    OPTION     option_table.h
    CC         postcar/soc/intel/common/block/cpu/car/exit_car.o
    HOSTCC     cbfstool/rmodtool (link)
python: can't open file '/home/bcran/src/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py': [Errno 2] No such file or directory
    HOSTCC     cbfstool/cbfstool.o
    HOSTCC     cbfstool/cbfs_image.o
    HOSTCC     cbfstool/cbfs-mkstage.o
make: *** [src/drivers/intel/fsp2_0/Makefile.inc:91: build/Fsp_M.fd] Error 2
make: *** Waiting for unfinished jobs....
```

### Steps to reproduce
```
git clone https://github.com/system76/firmware-open
cd firmware-open
./scripts/deps.sh
./scripts/build.sh addw3
```

### Expected behavior
The build should succeed.

### Actual behavior

The build fails.

### Additional info

```
bcran@system76-pc:~/src/firmware-open$ cat .git/modules/coreboot/config 
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	worktree = ../../../coreboot
[remote "origin"]
	url = https://github.com/system76/coreboot.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "system76"]
	remote = origin
	merge = refs/heads/system76
[submodule "3rdparty/amd_blobs"]
	active = true
	url = https://review.coreboot.org/amd_blobs
	update = none
[submodule "arm-trusted-firmware"]
	active = true
	url = https://review.coreboot.org/arm-trusted-firmware.git
[submodule "3rdparty/blobs"]
	active = true
	url = https://review.coreboot.org/blobs.git
	update = none
[submodule "3rdparty/chromeec"]
	active = true
	url = https://review.coreboot.org/chrome-ec.git
[submodule "3rdparty/cmocka"]
	active = true
	url = https://review.coreboot.org/cmocka.git
	update = none
[submodule "3rdparty/ffs"]
	active = true
	url = https://review.coreboot.org/ffs.git
[submodule "3rdparty/fsp"]
	active = true
	url = https://review.coreboot.org/fsp.git
	update = none
[submodule "intel-microcode"]
	active = true
	url = https://review.coreboot.org/intel-microcode.git
	update = none
[submodule "3rdparty/intel-sec-tools"]
	active = true
	url = https://review.coreboot.org/9esec-security-tooling.git
[submodule "libgfxinit"]
	active = true
	url = https://review.coreboot.org/libgfxinit.git
[submodule "libhwbase"]
	active = true
	url = https://review.coreboot.org/libhwbase.git
[submodule "opensbi"]
	active = true
	url = https://review.coreboot.org/opensbi.git
[submodule "3rdparty/qc_blobs"]
	active = true
	url = https://review.coreboot.org/qc_blobs.git
	update = none
[submodule "3rdparty/stm"]
	active = true
	url = https://review.coreboot.org/STM
[submodule "vboot"]
	active = true
	url = https://review.coreboot.org/vboot.git
[submodule "util/goswid"]
	active = true
	url = https://review.coreboot.org/goswid.git
[submodule "util/nvidia-cbootimage"]
	active = true
	url = https://review.coreboot.org/nvidia-cbootimage.git
```

crawfxrd, 2023-05-20:
Did you run `./scripts/deps.sh`? If so, then run `./scripts/update.sh`.

bcran, 2023-05-20:
Thanks. Running `git submodule update --checkout` or ./scripts/update.sh fixes the problem and allows me to successfully build the firmware.

closed 2023-05-22 by leviport