system76/firmware-open #420: will lemp11 be getting a firmware update

JordanBoulan, 2023-05-24:
I just got my lemp11 a few months ago. It still doesn't support s3 sleep which is very frustrating.
Secure boot would be a nice feature for windows 11 eventually as well. 
I see in the changelog s3 has been fixed and secure boot has been added.

Please at least fix S3 sleep, that is a huge deal.

So just want to know if the lemp11 will eventually get an update with s3 sleep and secure boot support because I don't see a reason it couldn't support those features. 

Thanks.

leviport, 2023-05-24:
As you say, it has been added in this repo. Official releases are built from this repo and extensively QA tested to make sure nothing bad will happen when the new firmware is released into the wild. Building and flashing from this repo yourself is doable, but there are no guarantees that there will be no regressions or negative behavior since it hasn't had the same test coverage. 

We are planning another official release for lemp11, but we don't currently have a timeline on that. Hopefully we can get to it soon.

crawfxrd, 2023-05-31:
Most boards have updates to 2023-05-25_5608a8d staged and need to be approved before release.

closed 2023-05-31 by crawfxrd