system76/firmware-open #424: lemp10: Suspend fails

crawfxrd, 2023-06-06:
- Model: lemp10
- BIOS version: `darp9` branch
- OS: Pop!_OS 22.04
- Kernel: 6.2.6-76060206-generic

lemp10 does not suspend.

### Steps to reproduce

- `sudo systemctl suspend`

### Expected behavior

- The system enters S0ix
- Power LED blinks green

### Actual behavior

- Power LED remains solid green or turns off
- Cannot power off the machine by holding the power button

### Additional info

Only output from EC is:

```
VWCTRL1 80
VWIDX47 11
VWCTRL1 80
VWIDX47 10
VWCTRL1 80
VWIDX47 11
VWCTRL1 80
VWIDX47 10
VWCTRL1 80
VWIDX47 11
```

crawfxrd, 2023-06-08:
```
[  384.033665] PM: suspend entry (s2idle)
[  384.043761] Filesystems sync: 0.010 seconds
[  384.045885] Freezing user space processes
[  384.060710] Freezing user space processes completed (elapsed 0.014 seconds)
[  384.060717] OOM killer disabled.
[  384.060719] Freezing remaining freezable tasks
[  384.061969] Freezing remaining freezable tasks completed (elapsed 0.001 seconds)
[  384.061977] printk: Suspending console(s) (use no_console_suspend to debug)
[  384.303111] ACPI: EC: interrupt blocked
[  395.559605] ACPI: EC: interrupt unblocked
[  398.347532] ACPI Error: AE_TIME, Returned by Handler for [EmbeddedControl] (20221020/evregion-300)
[  398.347537] ACPI Error: Timeout from EC hardware or EC device driver (20221020/evregion-310)

[  398.347548] No Local Variables are initialized for Method [_PSW]

[  398.347549] Initialized Arguments for Method [_PSW]:  (1 arguments defined for method invocation)
[  398.347550]   Arg0:   000000009c179d8f <Obj>           Integer 0000000000000000

[  398.347557] ACPI Error: Aborting method \_SB.LID0._PSW due to previous error (AE_TIME) (20221020/psparse-529)
```

etc.

justanotherrandomuser386, 2023-06-11:
- Model: lemp10
- BIOS version:  2023-05-25_5608a8d
- EC version: 2023-05-25_5608a8d
- OS: Fedora 38
- Kernel: 6.3.6-200.fc38.x86_64

Got same issue after flashing recent firmware update. But problem is much more severe than suspend failure. Actual bug is that after failed attempt to suspend (think of closing the lid on working laptop, pretty common scenario with mobile computers) you will get unusable laptop unresponsive to power button (long/short/eternal press) so you can't turn it on or off until you disassemble it and unplug the battery. After that laptop works as usual. At least this time you do not need an external programmer to bring machine back to life, yay!

curiousercreative, 2023-06-12:
Also affected after flashing updated firmware:
- galp5
- latest Pop!_OS

curiousercreative, 2023-06-12:
@jacobgkau yes and to be more precise, I flashed firmware myself. I haven't got back into the system yet to say for certain, but I believe I flashed 2022-11-21_b337ac6 with a modified coreboot (rebased to system76 branch) and modified EC (rebased to master).

curiousercreative, 2023-06-12:
You probably already figured out as much, but it seemed to be EC that was responsible for the failure. Once I rebased EC from 2022-11-21_b337ac6, all is good.

justanotherrandomuser386, 2023-06-13:
Yes, i just flashed mysefl, willigly taking all the risks. So no blaming anyone of system76 team for lack of QA. The main point was that 'does not suspend' is not the same as 'Cannot power off the machine by holding the power button'. Anyway, is there some kind of watchdog available in EC to reset the controller in case of unresponsive firmware?

crawfxrd, 2023-07-28:
```
VWIDX47 11
```

Is `HOST_C10`, now reported from [CB:73689](https://review.coreboot.org/c/coreboot/+/73689). Are we suppose to ack it or something?
I also see that is gets reset at `PLTRST#` and not `ESPI_RESET#`. Is that significant?

crawfxrd, 2023-07-31:
`GetTemp()` is being called while the CPU is in C10.

PR #444 (2023-08-01): Update EC and coreboot
Brings in the changes to resolve:

- Resolves: #78
- Resolves: #393
- Resolves: #424
closed 2023-08-01 by jackpot51