system76/firmware-open #437: Boot entries in NVRAM are frequently lost

marcin-rzeznicki, 2023-07-10:
- Model: oryp10
- BIOS version: 2023-06-08_36c78ea
- EC version: Flashed separately to modify fan curves and disable firmware security. Based on `dd555b9`
- OS: Arch
- Kernel: 6.4.2-arch1-1

Boot entries in NVRAM are lost randomly. Setting them manually via `efibootmgr` works randomly, requiring multiple reboots sometimes

### Steps to reproduce

Boot; reboot with some USB disks plugged in and an external monitor attached to DP (or HDMI, I guess, but it rarely boots) (I am not sure if this triggers the problem, but my observations happen to confirm that it usually happens under these conditions); after some time you will reboot to some PopOS stuff or whatever you have; issue `efibootmgr -v` from whatever you booted into and observe that your entries are gone; issue `efibootmgr --create ...`; reboot (and pray you're booted into your bootloader)

### Expected behavior

I always boot into what I want to boot

### Actual behavior

I spend more time in Arch live-disk and/or PopOS recovery something than I would like to

### Additional info

This started happening after I upgraded into `2023-06-08_36c78ea` - nothing of that sort had been happening before. I disabled firmware security because I thought that was the problem (boot process was hanging when it tried to display the screen to enter the digits every time I plugged in any external monitor, worked fine otherwise), but apparently not. The issue happens randomly from my point of view. Sometimes it works for days, sometimes I have to do it four times a day.

crawfxrd, 2023-09-05:
Likely either an issue with FaultTolerantWrite or variables being detected as corrupted when written. Probably need to cherry-pick https://github.com/MrChromebox/edk2/commit/d91161196b0e2fb4a11fdadc1c39e2457f617428.

Would require edk2 debug output to check what is causing SMMSTORE to be erased, but it's a fucking nightmare.

crawfxrd, 2023-09-05:
Steps to reproduce:

- Run `sudo bootctl install`
- Reboot
- Check `efibootmgr`

I'm seeing it happen on a dev unit anywhere between 1 and 3 reboots when booting on battery power.

crawfxrd, 2023-09-06:
oryp6 (CML-H) and oryp8 (TGL-H) appear to be unaffected.

marcin-rzeznicki, 2023-09-06:
Thanks for finally looking into it. Looking forward to a solution, 'cause it's unnerving

crawfxrd, 2023-09-06:
A couple things I see on darp8:

CSME is in an invalid state:

```
[DEBUG]  ME: Current Working State       : 4
[DEBUG]  ME: Current Operation State     : 1
[DEBUG]  ME: Current Operation Mode      : 3
[DEBUG]  ME: Error Code                  : 2
```

Setting up SMI failed:

```
[DEBUG]  FMAP: area SMMSTORE found @ 1010000 (262144 bytes)
[DEBUG]  smm store: 4 # blocks with size 0x10000
[INFO ]  SMMSTORE: Setting up SMI handler
[ERROR]  SMMSTORE: Failed to install com buffer
```

crawfxrd, 2023-09-07:
The APM SMI is failing with 0x4ed; I don't know what that means. Retrying it seems sufficient to get it to work though.

ilikenwf, 2023-09-07:
Interestingly, I think it may be related to setup mode. Once I reran sbctl it showed that secureboot was off and in setup mode...it seems to be sticking this time after re-enrolling, yet again. Not sure what was different.

closed 2023-09-08 by crawfxrd

crawfxrd, 2023-09-09:
2023-09-08_42bf7a6 staged for release.

marcin-rzeznicki, 2023-09-09:
Thanks, while we're at it - could you please tell your support not to convince people to ship their machines halfway around the world if they encounter similar problems, based on "hardware issues"? It would help others. Thank you in advance.

ilikenwf, 2023-09-09:
I don't work for sys76. I'm just a customer.

marcin-rzeznicki, 2023-09-10:
> I don't work for sys76. I'm just a customer.

Sorry, that message was meant for @crawfxrd

thomas-zimmerman, 2023-09-11:
@marcin-rzeznicki Support team just spoke about this bug today. We also don't like to ship machines halfway around the world for something that can be fixed in place.

ilikenwf, 2023-09-11:
I'm not really part of this particular issue but maybe you can train and provide programming tools to certified techs in a few other nations as contractors to fix such issues instead?

techsy730, 2023-09-16:
Just reporting that I have been encountering the same thing on a 2024 Bonobo (bonw15) on firmware version 2023-07-10_0e4a64a

Just in case this wasn't already known to affect this pair of model and firmware version.

marcin-rzeznicki, 2023-10-04:
> 2023-09-08_42bf7a6 staged for release.

Any idea when it's going to be actually released?

marcin-rzeznicki, 2023-11-14:
Ping @crawfxrd - it's been a while

crawfxrd, 2023-11-14:
https://github.com/system76/firmware-open/issues/340