system76/firmware-open #442: bonw14: Graphics don't work pre-OS

crawfxrd, 2023-07-28:
- Model: bonw14
- BIOS version:
- EC version:
- OS: Pop!_OS 22.04
- Kernel: 6.2.6-76060206-generic

Graphics output does not work until the OS loads.

### Steps to reproduce

- Boot the system

### Expected behavior

- Boot splash is visible
- Plymouth disk decryption screen is visible

### Actual behavior

- No graphics output until OS loads

### Additional info

bonw14 is uniquely the only system that does not use Intel graphics. The eDP is connected to the dGPU.

crawfxrd, 2023-07-28:
I will need `cbmem -c` output from current release firmware.

I have the output from `2023-07-06_8e24a60`, but I guess it would be good to get it on a version with microcode loading fixed (200ddc00875e or newer).

crawfxrd, 2023-08-09:
This may the same issue as external displays on dGPU ports non working other models.

jacobgkau, 2023-09-01:
> I will need `cbmem -c` output from current release firmware.

Here is that, from `2021-07-20_93c2809`: [bonw14-93c2809-cbmem-c.txt](https://github.com/system76/firmware-open/files/12495454/bonw14-93c2809-cbmem-c.txt)

crawfxrd, 2023-12-19:
Here is a build with just edk2 and firmware-setup (because of build errors) updated:

- [bonw14-edk2.zip](https://github.com/system76/firmware-open/files/13720332/bonw14-edk2.zip)

This is based on fb9d759338c5, which is the last commit before coreboot was updated to 4.15.

This may break boot, so only flash this if you're willing to suffer externally flashing.
Otherwise I will need a unit so only I have to cry.

leviport, 2023-12-19:
I will try this soon. I don't have my machine with me at the office today, but I'll set it up in a test bench config at home. I have a set of flashing tools at home in case things become unalive.

crawfxrd, 2023-12-21:
Option ROM scanning in edk2 doesn't work anymore. `PciGetPciRom` is never run.

crawfxrd, 2023-12-21:
Here is a build based on 6f1e65308e29 with edk2 downgraded to system76-stable202002.

- [bonw14-downgrade-edk2.zip](https://github.com/system76/firmware-open/files/13746877/bonw14-downgrade-edk2.zip)

leviport, 2023-12-26:
That build did the trick! I now get the System76 splash screen at boot, can navigate firmware menus, and see the Plymouth decrypt prompt.

jens-siebert, 2024-02-17:
Hey everyone,

I was very pleased to see that a firmware update for the bonw14 is being worked on, but I have the impression that progress here has stalled...

I really don't want to be a pain, but as an owner of such a machine I'm of course very interested in this topic. Is there any progress to report here? Are there any plans for when we will see the updated firmware for the bonw14?

Thanks a lot and keep up the great work!

crawfxrd, 2024-02-20:
No progress. The problems:

- I don't have a bonw14 to debug this issue
- Even if I did, I don't want to disassemble and reassemble the unit in order to flash it
- The issue is apparently in edk2, and I don't want to debug that shit