system76/firmware-open #443: 404 error while installing dependencies

Vitamin-X, 2023-07-29:
when running ./scripts/deps.sh the installer fails at
nasm-2.15.05.tar.bz2 (downloading from https://www.nasm.us/pub/nasm/releasebuilds/2.15.05/nasm-2.15.05.tar.bz2 )...   0%
because that site didn't exist,

i would like to change the link to 
https://anaconda.org/conda-forge/nasm/2.15.05/download/linux-64/nasm-2.15.05-h7f98852_0.tar.bz2
 but i cant fine the file for it, or just insert the file into the right folder.
 
 can someone give me a little advise here? or just wait until the site is back again?

crawfxrd, 2023-07-29:
This is part of the coreboot toolchain.

You can wait for the site to come back up, or download the tarball from somewhere else like archive.org and put it in `coreboot/util/crossgcc/tarballs/`.

Vitamin-X, 2023-07-29:
now im getting a hash mismatch

* nasm-2.15.05.tar.bz2 (cached)... hash mismatch:
expected (known) hash: c0869e9ac854140bf99c989a711102ce6367243f
calculated hash of downloaded file: 3be3ac945f23226da32de7186aa9291e0b4218b7

closed 2023-07-29 by Vitamin-X