system76/firmware-open #453: Error installing dependencies to build the firmware (downloading acpica-unix2-20220331.tar.gz fails for unknown reasons)

k4l1mi5t, 2023-08-04:
- Model: lemp10
- BIOS version: 2022-11-21_b337ac6
- EC version: 2022-11-21_b337ac6
- OS: Pop!_OS 22.04 LTS
- Kernel: 6.4.6-76060406-generic

I'm trying to adjust the fancurve on my lemur as it's a bit to agressive for my taste, spinning up for a couple of seconds and stopping again, over and over. 

I run into an error installing the dependencies to build the firmware.

### Steps to reproduce

git clone https://github.com/system76/firmware-open.git
cd firmware-open
git checkout b337ac6
git submodule init
git submodule update
sudo ./scripts/deps.sh

Installing system build dependencies
Hit:1 https://apt.releases.hashicorp.com jammy InRelease
Hit:2 https://ppa.launchpadcontent.net/appimagelauncher-team/stable/ubuntu jammy InRelease
Hit:3 http://packages.microsoft.com/repos/code stable InRelease
Ign:4 https://storage.googleapis.com/download.dartlang.org/linux/debian stable InRelease
Hit:5 https://dl.google.com/linux/chrome/deb stable InRelease
Hit:6 https://storage.googleapis.com/download.dartlang.org/linux/debian stable Release
Hit:8 http://apt.pop-os.org/proprietary jammy InRelease
Hit:9 http://apt.pop-os.org/release jammy InRelease
Hit:10 http://apt.pop-os.org/ubuntu jammy InRelease
Hit:11 http://apt.pop-os.org/ubuntu jammy-security InRelease
Hit:12 http://apt.pop-os.org/ubuntu jammy-updates InRelease
Hit:13 http://apt.pop-os.org/ubuntu jammy-backports InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
libudev-dev is already the newest version (249.11-0ubuntu3.9pop0~1689262825~22.04~3be1d98).
bison is already the newest version (2:3.8.2+dfsg-1build1).
build-essential is already the newest version (12.9ubuntu3).
dosfstools is already the newest version (4.2-1build3).
flex is already the newest version (2.6.4-8build2).
msr-tools is already the newest version (1.3-4).
mtools is already the newest version (4.0.33-1+really4.0.32-1build1).
parted is already the newest version (3.4-2build1).
python-is-python3 is already the newest version (3.9.2-2).
uuid-dev is already the newest version (2.37.2-4ubuntu3).
ccache is already the newest version (4.5.1-1).
devmem2 is already the newest version (0.0-0ubuntu2).
flashrom is already the newest version (1.2-5build1).
gnat is already the newest version (10.1ubuntu1).
curl is already the newest version (7.81.0-1ubuntu1.13).
libncurses-dev is already the newest version (6.3-2ubuntu0.1).
python3-distutils is already the newest version (3.10.8-1~22.04).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9.2).
git-lfs is already the newest version (3.0.2-1ubuntu0.2).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Installing GIT LFS hooks
Updated git hooks.
Git LFS initialized.
Downloading GIT LFS artifacts
Initializing submodules
Skipping submodule '../../coreboot/'
Skipping submodule '../../coreboot/'
Skipping submodule '../../coreboot/'
Skipping submodule '../../coreboot/'
Skipping submodule '../../coreboot/'
Skipping submodule '../../coreboot/'
Installing coreboot commit hook
Installing Rust
[... 145 lines cut ...]
Unpacked and patched ... ok
Building packages ...
Skipping NASM v2.15.05 for host as it is already built
Packages built ... ok
Copied EDK2 tools template ... ok
Cleaning up temporary files... ok

You can now run NASM from /home/obiwan/gits/firmware-open/coreboot/util/crossgcc/xgcc.
Welcome to the coreboot cross toolchain builder v2022-07-30_ae157d38e3

Building toolchain using 8 thread(s).

Downloading and verifying tarballs ...
 * acpica-unix2-20220331.tar.gz (downloading from https://acpica.org/sites/acpica/files/acpica-unix2-20220331.tar.gz)...   0%... Failed to download acpica-unix2-20220331.tar.gz.
make[1]: *** [Makefile:23: build_iasl] Error 1
make: *** [util/crossgcc/Makefile.inc:30: crossgcc-i386] Error 2
Failed to install dependencies!

### Expected behavior

Running the deps.sh script will successfully install needed dependencies to build the firmware.

### Actual behavior

Script errors out trying to download acpica-unix2-20220331.tar.gz

crawfxrd, 2023-08-04:
Intel has removed the acpica.org site. It now just redirects to ACPICA page intel.com. You need to use a newer version of coreboot, or download the archive from somewhere else (archive.org, GitHub).

closed 2023-08-05 by crawfxrd

k4l1mi5t, 2023-08-05:
Thank you, that worked out. Only thing is if I don't run the script as root, it's not able to lock the directory /var/lib/apt/lists/. See the following error:

Installing EC dependencies
~/gits/firmware-open/ec ~/gits/firmware-open
Installing system build dependencies
Reading package lists... Done
E: Could not get lock /var/lib/apt/lists/lock. It is held by process 1589 (packagekitd)
N: Be aware that removing the lock file is not a solution and may break your system.
E: Unable to lock directory /var/lib/apt/lists/
Failed to install dependencies!
Failed to install dependencies!

You can leave it as closed as my issue is resolved.