system76/firmware-open #460: lemp10: keyboard is ignored when opening UEFI firmware settings

ethindp, 2023-08-13:
- Model: lemp10
- BIOS version: 2022-11-21
- EC version: 2022-11-21
- OS: Windows and FreeBSD

When opening firmware settings, the keyboard gets ignored when trying to navigate around (that is, keyboard input does nothing).

### Steps to reproduce

I'm honestly unsure how to reproduce this. I do have Windows installed, so maybe that's a contributing factor. But whenever I try to boot into firmware settings via ESC, most of the time I can't navigate around -- keyboard input does absolutely nothing. But if I tell Windows to boot into firmware settings, it works, at least all the times I've tried it. So maybe this is something Windows is doing?

### Expected behavior

I should be able to navigate around just fine.

### Actual behavior

All keyboard input is ignored.

### Additional info

I have Windows 10 22H2 (AMD64) build 19045.3324 installed. (from the behavior I've observed, any Windows 10/11 install will work.)

crawfxrd, 2023-08-13:
Should be resolved by https://github.com/system76/ec/pull/377.

eblanshey, 2023-08-30:
@crawfxrd we just had a scary situation here. We received a notice for a firmware update, so after rebooting it booted into the "confirm the code below" screen... but there was no response when trying to type anything. Completely frozen. Couldn't boot into recovery, either, as pressing escape or space or anything had no effect. I thought the firmware update had bricked the laptop. After some searching I discovered this keyboard issue, and I got the idea to plug in an external keyboard in case the built-in one was disabled, and behold, it worked. Very scary.

I wonder if the firmware update was fixing this particular keyboard issue, causing a catch-22 😛

crawfxrd, 2023-08-30:
An EC reset (power off an unplug the unit) will resolve the problem (put KBC in the correct state) until the new firmware is released.

crawfxrd, 2023-09-11:
Duplicate of #406

closed 2023-09-11 by crawfxrd