system76/firmware-open #468: Req: Push firmware updates

MrPenguin07, 2023-08-31:
Hi dev team.

First and foremost I want to stress that I see a lot of commits, a lot of great work being done here.

However all of this work is being withheld from we end user customers due to what?
Firmware releases are important to thoroughly go through QA, failed updates are a real problem if things are rushed.

However I have to be honest, I think S76 aren't dedicating enough resources to back porting firmware. I've seen multiple rounds of layoffs, yet development on other projects such as pop and cosmic seem to receive the lions share of dev budget.

A few months ago, we were given the announcement for firmware updates being back ported, and the git branches haven't changed since.

My latest device is darp8, and fixes and features waiting to be released are rather critical... Unless I'm mistaken the commits I've seen allow disabling of ME, enable secure boot, suspend si0x fixes (despite claims, the workaround implemented is not great) and some more.

This is a plea for more QA to get these out please S76.

crawfxrd, 2023-08-31:
#340

closed 2023-08-31 by crawfxrd

leviport, 2023-08-31:
We're doing what we can with what we have, apologies if you find it insufficient. Thank you for your patience while we work our way through these updates. I just so happen to be working on darp8 right now, so you should be seeing that update soon.

jackpot51, 2023-09-01:
The darp8 firmware has been updated