system76/firmware-open #470: serw13: Occasionally Boot Entry for Grub Is Randomly Lost, An Issue When Secureboot Enabled

ilikenwf, 2023-09-03:
- Model: serw13
- BIOS version: 2023-05-17_9560b2e
- EC version: 2023-05-17_9560b2e
- OS: Archlinux
- Kernel: Any kernel

Sometimes, between reboots, I inexplicably loose the Grub boot entry. This is a problem since I use my own secureboot keys, have an encrypted /boot, and am not using the shim. In addition, I don't install grub in portable mode, because I want to keep things locked down.

Whenever this happens I have to boot a live USB and re-add the Grub boot entry. It makes no sense. In addition, the firmware automatically disables secureboot when this happens, which is mind bogglingly stupid if we really are seeking to be secure - and while I realize evil maids are still a thing if they have a chip clip, why is there no option to password protect the firmware setup? 

I paid a large sum for this amazing hardware but am pretty upset with the software, because these are all things I can do if I build my own Coreboot for my old crusty thinkpads...

### Steps to reproduce

I'm not sure if any logical thing is causing this. I re-sign my bootloader and vmlinuz every time they change. I'm not sure if this, just reboots, or the act of plugging in flash drives (while booted into the OS) is causing this, but it is really really maddening.

### Expected behavior

Grub should not be lost from boot options, and it shouldn't be a requisite to have a grub install in portable mode (which makes the signature check irrelevant if SecureBoot is disabled),

ilikenwf, 2023-09-04:
Sorry for the tone, I was up all night working on this and a couple other things...

crawfxrd, 2023-09-05:
Duplicate of #437

closed 2023-09-05 by crawfxrd