system76/firmware-open #471: lemp12 key debounce too lax on 023-08-18_a8dd6c2

sararocks, 2023-09-05:
Customer is having an issue with keys typing too many times during a normal use case. 
"I can't type on this keeyboardd. I swear I'm not joking, look at the mess above/the two issues. This is a nightmare to use and, as stated, I only have the laptop for 24h so far.r (<- yet another doubule <- yet another double key)." 
(customer left the errors to illustrate his point) 

Another issue, sometimes the typed letter repeats after the fact: "because it looks like there's a delayed/deeebounceed key showing up AFTER something else. So rarely, sometitmes OH THERE IT IS."
### Steps to reproduce

### Expected behavior

### Actual behavior

### Additional info

Original customer message:
"I'll intentionally NOT correct any errors that aren't mine, i.e. created by the keyboard. One example is already in the subject: It seems that the debouncing (?) is quite bad. In the few hours I have this laptop - it arrived only yesterday - I have run into this problem quite a lot / regularly and iit's especially bad if I try typing a password or something important and don't notice the error (think typing my own name for an officiial document or giving my address for a delivery etc))
So far, looking up, I can see 4 of these instances [ee, ii, ii, ))] happening in this short tticket (there it was again).

There's another keyboard issue that is harder to trigger, so I haven't produced it in this tiny amount of ticket text. It's EXTRA weirrd, because it looks like there's a delayed/deeebounceed key showing up AFTER something else. So rarely, sometitmes OH THERE IT IS. Believe me that I pressed T once in sometimes. It regisitered (SAME ERROR HERE with i) twice, BUT the second one seemingly came out of nowhere.

In short: I can't type on this keeyboardd. I swear I'm not joking, look at the mess above/the two issues. This is a nightmare to use and, as stated, I only have the laptop for 24h so far.r (<- yet another doubule <- yet another double key)."

jacobgkau, 2023-09-05:
It sounds like the customer just received the laptop, and didn't start experiencing the issue after a firmware update. Is it possible this could be a faulty keyboard or bad ribbon connection instead of a firmware issue?

crawfxrd, 2023-09-05:
darp8, darp9, galp7, and lemp9 are also on 2023-08-18_a8dd6c2. Has this been reported against any of them?

thekingrat, 2023-09-07:
Can confirm I have this problem on darp8 after installing the firmware update.

jackpot51, 2023-09-07:
@crawfxrd we may want to adjust debounce time, perhaps to something like 10ms.

closed 2023-09-08 by crawfxrd