system76/firmware-open #473: bonw15 firmware very slow and sometimes freezes

Raikiri, 2023-09-06:
- Model: bonw15
- BIOS version: 2023-07-10_0e4a64a
- EC version: 
- OS: N/A
- Kernel: N/A

When I enter the firmware interface (by pressing esc during booting), the up/down buttons have a very long delay, about 1 to 2 seconds. However, sometimes the firmware straight up freezes either immediately or after pushing the said up/down buttons a couple times. Sometimes it unfreezes in 5-10 seconds, but most of the time when it happens it freezes for good, fans idle.

### Steps to reproduce

Press esc during boot

### Expected behavior

I don't expect the firmware to freeze

### Actual behavior

The firmware freezes

Raikiri, 2023-09-08:
I also had a freeze yesterday in windows 10 while not doing anything. No prior hiccups, no other symptoms, just a freeze while browsing the settings menu (did not freeze again when I rebooted and browsed the same page again).

My old laptop (non-s76) had a similar issue that it froze specifically when _almost_ nothing was happening, turns out that was due to one of the lower-power C-states causing some sort of failure. After months of investigation, the workaround for that laptop was to disable some of the C-states. Maybe it's related, maybe it's not.

Raikiri, 2023-09-11:
@leviport have you tried it? could you repro it?

leviport, 2023-09-11:
Haven't had a chance to yet. My days are very full lately, sorry. I will let you know once I get a chance.

Raikiri, 2023-09-14:
Further testing revealed that I could no longer repro this issue. My suspicion is that it froze when I had one specific USB stick inserted. That USB stick has eventually proven to be faulty (weird read errors, later stopped being detected in the OS), but can it lead to a freeze?

I don't remember whether the USB stick was in the laptop when it froze in windows, but I suspect that it might have been.

Testing: I went into the firmware menu, kept navigating it for a couple minutes and rebooted ~5 times, no freezes. I also did have a usb stick inserted as well, but a different one (not the faulty one).

closed 2023-09-14 by leviport