system76/firmware-open #477: firmware flashing for fan speeds has no effect

John-Bronson, 2023-09-14:
- Model: gaze17-3060-b
- BIOS version: 2022-06-07_090f9e0
- EC version: 
- OS: Pop!_OS 22.04 LTS x86_64
- Kernel: 6.4.6-76060406-generic

My gaze17-3060-b has super loud fans under even moderate load. They also come on suddenly and oscillate up and down rapidly. I've tried to flash the firmware using the guide found here (which seems to match similar procedures I've seen in forums and here on github): https://devimalplanet.com/guide-how-to-adjust-the-fan-curve-on-system76-laptops

When I flash the firmware and reboot, the fans seem to behave exactly as they did before. And there's no way I know of to actually read the settings to see if they've taken.

I contacted Pop OS technical support and they sent me here. I hope this is enough information for someone to help, as I honestly don't know what else to do about this thing.

(Yes, I know about running on Battery Saver. I run it that way 24/7. Every time I boot this thing up I set it to Battery Saver and then turn up the brightness back up on the screen, even though this thing is always plugged in. This makes it possible to attend video conferences, but it also makes the computer slower)

ahoneybun, 2023-09-27:
Those steps do not work for systems with the new firmware security. The steps would be like this:

git clone https://github.com/system76/firmware-open
cd firmware-open
git checkout *commit*

Change the commit to match the released firmware for your model (090f9e0 in this case)

git submodule update --init --recursive
./scripts/deps.sh
source ~/.cargo/env

Then edit the board file for your system and change the fan curve values:

gedit ec/src/board/system76/gaze17-3060-b/board.mk

Once that is done save and close Gedit then run these commands to build and flash the custom firmware:

./scripts/build.sh gaze17-3060-b
./scripts/flash.sh gaze17-3060-b
sudo systemctl reboot

BTBlueSkies, 2024-02-24:
The steps above by ahoneybun are not working on this day, feb 23, 2024. I have a gazel6-3050, and when following those steps the first problem is that the acpica repository links have changed and one has to manually grab the acpica file and drop it into the tarball files for the build tree. The second problem is that after this manual fix to the build tree, the build then fails with some $arch variable not found/set/etc. Then, deciding to grab the latest version of the github repo for the firmware-open and running the ./scripts/install-deps.sh (on a fresh install to 22.04 with all updates on the gazel6).. this command fails with some github error asking me git ocnfig --global user.name .... etc. 

Just an update on my experience this last few hours. I think the build tree is fragile?

ahoneybun, 2024-02-24:
> The steps above by ahoneybun are not working on this day, feb 23, 2024. I have a gazel6-3050, and when following those steps the first problem is that the acpica repository links have changed and one has to manually grab the acpica file and drop it into the tarball files for the build tree. The second problem is that after this manual fix to the build tree, the build then fails with some $arch variable not found/set/etc. Then, deciding to grab the latest version of the github repo for the firmware-open and running the ./scripts/install-deps.sh (on a fresh install to 22.04 with all updates on the gazel6).. this command fails with some github error asking me git config --global user.name .... etc.
> 
> Just an update on my experience this last few hours. I think the build tree is fragile?

It is because of the new security process where you need to enter a code to flash firmware. 

https://github.com/system76/firmware-open/blob/master/FEATURES.md#firmware-security-system

Logic-Beach, 2024-02-25:
Is there any hope for adjusting the fan curve on this new secure firmware? I tried another [method found on reddit](https://devimalplanet.com/guide-how-to-adjust-the-fan-curve-on-system76-laptops) but unfortunately this fails when running  `make_internal`. It's discussed in more detail in this other [GH issue](https://github.com/system76/firmware-open/issues/250)

I am using a serw13 and when running anything that slightly stresses the CPU it's 5000 RPM fan is very annoying. I feel like it'd be better off kicking on slower at lower temps and never getting to 100% speed.

Logic-Beach, 2024-02-25:
UDPATE: I just tried the steps provided by @ahoneybun on [reddit](https://www.reddit.com/r/System76/comments/16sg78r/comment/k29x2n5/) (in the comments), and it flashes the firmware fine. Thanks for that at least that's working.

But sadly I have to confirm that there is no effect on the actual fan curve after editing the board.mk file. I tested scarily low values to confirm. No matter the values the fans shoot to 6000 RPM like normal.

Can we get any further guidance from the system76 team on this?

crawfxrd, 2024-02-25:
Post your diff.

Logic-Beach, 2024-02-26:
Edited: 
[image]

when I git-pulled master, I saw the same file... not sure what I did, but I must edited there too.. I am newish to git

crawfxrd, 2024-07-05:
Not reproducible.

Flashing should be done with `./scripts/flash.sh` to use firmware-update.

closed 2024-07-05 by crawfxrd