system76/firmware-open #480: addw2 RAM speed limited to 2933 MHz (documentation says 3200 MHz)

jacobgkau, 2023-09-20:
- Model:  addw2
- BIOS version:  2023-09-08_42bf7a6 & 2022-11-21_b773ac6.
- EC version:  2023-09-08_42bf7a6 & 2022-11-21_b773ac6.
- OS:  Pop!_OS 22.04
- Kernel:  6.4.6-76060406

Tech docs list the addw2 as supporting up to 3200 MHz RAM, but it only runs at up to 2933 MHz (tested with 1x MNSO48E82-3200E in both slots individually and 2x Neo Forza NMSO432F82-3200EA00).

### Steps to reproduce

Boot the machine & run `sudo dmidecode -t 17`.

### Expected behavior

RAM runs at 3200 MHz.

### Actual behavior

RAM runs at 2933 MHz.

### Additional info

I can't seem to find the upstream Clevo manual for this system right now. I see in the addw1 Clevo manual that it was limited to 2666 MHz. If it turns out this is expected behavior, then I'll transfer this issue to the tech docs repo and fix it there.

This was discovered while testing the latest upcoming firmware update (2023-09-08_42bf7a6), but is not a regression from the current release (2022-11-21_b773ac6).

jacobgkau, 2023-09-20:
Checking the Joshua reports, we did sell 3200 MHz RAM in this system.

leviport, 2023-09-20:
From looking at https://ark.intel.com/content/www/us/en/ark/products/202329.html, it looks like that might be a CPU limitation

[image]

jacobgkau, 2023-09-20:
This system only had one CPU option, so not sure why we would have sold 3200 MHz in that case. Looks like 138 orders total (split between 2x16GB and 2x32GB), so it doesn't seem like a recordkeeping fluke.

jacobgkau, 2023-09-20:
Looking back in conversations, I do see where we stopped offering 3200 MHz as an option, but I was explicitly told that was due to supply-chain issues and not a technical limitation. I don't see any other past conversation about this internally.

crawfxrd, 2023-09-20:
Yes. From what I remember, overclocking was disabled because we got reports of the system locking up/crashing with 3200 MHz RAM.

jacobgkau, 2023-09-20:
That commit is from January of this year, the currently released firmware from November of last year also has the issue.

crawfxrd, 2023-09-20:
Timestamp doesn't matter because I cherry pick all the commits to a new base for the coreboot release updates.

Here is the same commit from when we were on 4.16 (`system76-4.16` branch):

- https://github.com/system76/coreboot/commit/8c4eeafcfe6ce68e2594bc8bc029195083bb37b8

Here is the upstream MR with a better description:

- https://review.coreboot.org/c/coreboot/+/67837

jacobgkau, 2023-09-20:
Got it, I'll edit the docs to note that 3200 MHz is theoretically supported but there may be stability issues with higher than 2933.