system76/firmware-open #481: Keyboard unresponsive after `lemp10` firmware update

AaronFeickert, 2023-10-08:
- Model: `lemp10`
- BIOS version: `2023-09-08_42bf7a6`
- EC version: n/a
- OS: `Pop!_OS 22.04 LTS`
- Kernel: `6.5.4-76060504-generic`

After the latest firmware update, leaving the laptop locked for several minutes (presumably entering some kind of suspend state) leaves the keyboard permanently unresponsive and the trackpad/mouse briefly unresponsive.

### Steps to reproduce

Lock the screen and wait for the display to turn off. Press a key to active the display and return to the unlock prompt. The keyboard and trackpad work as expected.

Lock the screen and walk away for a while. (The laptop presumably enters a suspend state, despite this not being specified in the power settings.) Press a key to activate the display and return to the unlock prompt. The keyboard is permanently unresponsive. The trackpad is sometimes briefly unresponsive. The only way to restore functionality is a hard reset via the power button.

If Bluetooth is on and a keyboard is connected, the Bluetooth keyboard works as expected.

### Expected behavior

The keyboard and trackpad should accept input.

### Actual behavior

The built-in keyboard does not accept input (but does function to activate the display). The trackpad sometimes briefly does not accept input. If a Bluetooth keyboard is connected, it accepts input.

### Additional info

This behavior only began after a firmware update.

AaronFeickert, 2023-10-08:
If the laptop is manually put into suspend mode (for example, by closing the lid) subsequent to the infuriating situation described above, the screen is unable to be activated at all.

crawfxrd, 2023-10-09:
Reproducible on my lemp10 with system76/ec@8ba23f5.

```
$ ./scripts/ectool.sh info
board: system76/lemp10
version: 2023-07-31_8ba23f5
```

When reproduced, EC becomes unresponsive:

```
$ ./scripts/ectool.sh info
failed to connect to EC: WouldBlock
```