system76/firmware-open #491: lemp10 not entering cstates lower than C2 while idling

leviport, 2023-10-16:
- Model: lemp10
- BIOS version: `2023-10-13_5d997d1`
- EC version: `2023-10-13_5d997d1`
- OS: Pop!\_OS 22.04
- Kernel: 6.5.4

When letting lemp10 sit idle and watching the cstates in the "Idle stats" tab of `powertop`, all but C2 stay at 0.0%. 

It does go into C10 while suspended though.

### Steps to reproduce

- Install `powertop` 
- Run `sudo powertop`, tab to the "Idle stats" page
- Let the machine sit idle for a few minutes
- All Cstates besides C2 should read `0.0%`

### Expected behavior

Cstates below C2 should be reachable

### Actual behavior

Only C2 is reachable while the machine idles

### Additional info
Tunables tab:
```
   Bad           Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Shared SRAM
   Bad           Runtime PM for PCI Device Intel Corporation Tiger Lake-LP SPI Controller
   Bad           Runtime PM for PCI Device Intel Corporation Wi-Fi 6 AX201
   Bad           Runtime PM for PCI Device Intel Corporation TigerLake-LP Dynamic Tuning Processor Participant
   Bad           Runtime PM for PCI Device Intel Corporation Tiger Lake-LP LPC Controller                               
   Bad           Runtime PM for PCI Device Intel Corporation GNA Scoring Accelerator module
   Bad           Runtime PM for PCI Device O2 Micro, Inc. SD/MMC Card Reader Controller
   Good          Bluetooth device interface status
   Good          NMI watchdog should be turned off
   Good          Enable Audio codec power management
   Good          VM writeback timeout
   Good          Runtime PM for I2C Adapter i2c-10 (i915 gmbus tc4)
   Good          Runtime PM for I2C Adapter i2c-7 (i915 gmbus tc1)
   Good          Runtime PM for I2C Adapter i2c-4 (i915 gmbus dpa)
   Good          Runtime PM for I2C Adapter i2c-6 (i915 gmbus dpc)
   Good          Runtime PM for I2C Adapter i2c-9 (i915 gmbus tc3)
   Good          Runtime PM for I2C Adapter i2c-0 (SMBus I801 adapter at efa0)
   Good          Runtime PM for I2C Adapter i2c-8 (i915 gmbus tc2)
   Good          Runtime PM for I2C Adapter i2c-1 (Synopsys DesignWare I2C adapter)
   Good          Runtime PM for I2C Adapter i2c-3 (Synopsys DesignWare I2C adapter)
   Good          Runtime PM for I2C Adapter i2c-5 (i915 gmbus dpb)
   Good          Autosuspend for USB device xHCI Host Controller [usb1]
   Good          Autosuspend for USB device xHCI Host Controller [usb2]
   Good          Runtime PM for I2C Adapter i2c-12 (i915 gmbus tc6)
   Good          Runtime PM for I2C Adapter i2c-2 (Synopsys DesignWare I2C adapter)
   Good          Autosuspend for unknown USB device 3-10 (8087:0026)
   Good          Autosuspend for USB device Chicony USB2.0 Camera [CHICONY]
   Good          Runtime PM for I2C Adapter i2c-11 (i915 gmbus tc5)
   Good          Autosuspend for USB device xHCI Host Controller [usb3]
   Good          Autosuspend for USB device xHCI Host Controller [usb4]
   Good          Runtime PM for PCI Device Intel Corporation Device a0c7
   Good          Runtime PM for PCI Device Intel Corporation Tigerlake PCH-LP PCI Express Root Port #6
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Serial IO I2C Controller #1
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Thunderbolt 4 USB Controller
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Management Engine Interface
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Serial IO I2C Controller #4
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP SMBus Controller
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Thunderbolt 4 PCI Express Root Port #0
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Thunderbolt 4 NHI #0
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller
   Good          Runtime PM for PCI Device Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics]
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller
   Good          Runtime PM for PCI Device Intel Corporation Tiger Lake-LP PCI Express Root Port #9
   Good          Runtime PM for PCI Device Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
   Good          Runtime PM for PCI Device Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers
```

Toggling all of the Bad's to Good doesn't seem to affect anything, and returning to the Tunables tab shows that they've all been switched back to Bad's by themselves.

crawfxrd, 2023-10-16:
> Runtime PM for PCI Device O2 Micro, Inc. SD/MMC Card Reader Controller

Okay, I'm just going to enable the driver on everything since we can't trust Clevo to swap the part.

leviport, 2023-10-16:
Solved by https://github.com/system76/firmware-open/pull/492

closed 2023-10-16 by leviport