system76/firmware-open #497: darp9: usb3 devices not detected on thunderbolt port

karterfone, 2023-10-25:
- Model: darp9
- BIOS version: 2023-08-18_a8dd6c2
- EC version: 2023-08-18_a8dd6c2
- OS: Ubuntu 22.04
- Kernel: 6.5.4-76060504-generic

USB3 devices are not detected by the Thunderbolt port, either through a dock, or directly connected to the TB4 port. The adjacent USB4 port is not affected. When a Thunderbolt dock is plugged in, external monitors and USB2 devices work normally, and `boltctl` shows a Thunderbolt dock connected and authenticated. Tested with two different TB4 docks and a separate Dell Latitude with TB4 as a control. The Latitude had no issues using any ports or devices on the docks.

### Steps to reproduce

Plug a USB3 device into the darp9 Thunderbolt port.

### Expected behavior

USB3 device is detected and visible to the OS.

### Actual behavior

USB3 devices are never detected. No error messages regarding USB3 connections show up in the system logs.

### Additional info

`lsusb -t` shows USB2 devices being attached to one bus. Nothing USB3 shows up there. When switching to the USB4 port, USB3 devices show up on a separate bus. Other OSes all had the same result in testing. The darp9 was sent in for warranty service, wherein the motherboard was reportedly replaced, but with no effect. The problem has persisted since initial purchase.

ahoneybun, 2023-10-25:
Is this when the device is connected at boot or if you connect the device after the system is booted? It could be connected to this issue:

https://github.com/system76/firmware-open/issues/343

karterfone, 2023-10-26:
Possibly, but with a different issue. If the device is connected at boot, it is detected as a USB2 device. If plugged into the standard USB-C or -A port, it is detected as USB3. Otherwise, the behavior is unchanged.