system76/firmware-open #498: TPM reports it's locked when trying to clear

thomas-zimmerman, 2023-11-02:
- Model: galp7
- BIOS version: 2023-09-08_42bf7a6
- EC version: 2023-09-08_42bf7a6
- OS: Pop!OS 22.04
- Kernel: 6.5.6

Trying to clear the TPM with `tpm2_clear` we get a TPM error:
```
ERROR: esys:src/tss2-esys/api/Esys_Clear:c97:Esys_Clear() Esys Finish ErrorCode (0x00000921)
ERROR: Esys_Clear(0x921) - tpm:warn(2.0): authorizations for objects subject to DA protection are not allowed at this time because the TPM is in DA lockout mode
ERROR: Unable to run tpm2_clear
```

### Steps to reproduce

```
sudo apt install tpm2-tools
sudo tpm2_clear
```
### Expected behavior

We expect to have the TPM cleared for setting up new keys for LUKS or BitLocker use.

ahoneybun, 2023-11-02:
Running this on a lemp12 with firmware build 2023-09-08_42bf7a6 gives me this output:

```
WARNING:esys:src/tss2-esys/api/Esys_Clear.c:291:Esys_Clear_Finish() Received TPM Error 
ERROR:esys:src/tss2-esys/api/Esys_Clear.c:97:Esys_Clear() Esys Finish ErrorCode (0x0000098e) 
ERROR: Esys_Clear(0x98E) - tpm:session(1):the authorization HMAC check failed and DA counter incremented
ERROR: Unable to run tpm2_clear
```

ahoneybun, 2023-11-02:
If I run this command I get the lockout mode error like the customer:

```
tpm2_dictionarylockout --setup-parameters --max-tries=4294967295 --clear-lockout ```

sun2sirius, 2023-11-02:
My main working platform is gaze18, which originally faced this issue. I did a bunch of experimenting on it before I saw this, like I ran Win11, built/run open firmware, etc. I though maybe it got into this state in the process. Then I got the galp7 literally out of the box, brand new, and it had the same issue. I wonder if it is possible to get in touch with someone from TPM manufacturer, because I see some other issues that I cannot explain. Thanks!

duplexsystem, 2023-12-29:
Try `tpm2_clear -c platform`

sun2sirius, 2023-12-29:
Yes, "-c" was the magic switch - thank you!

TobiPeterG, 2024-08-03:
Hi there, I get this output:
```
sudo tpm2_clear -c platform
WARNING:esys:src/tss2-esys/api/Esys_Clear.c:291:Esys_Clear_Finish() Received TPM Error 
ERROR:esys:src/tss2-esys/api/Esys_Clear.c:97:Esys_Clear() Esys Finish ErrorCode (0x000009a2) 
ERROR: Esys_Clear(0x9A2) - tpm:session(1):authorization failure without DA implications
ERROR: Unable to run tpm2_clear
```
Did I do something wrong?

thomas-zimmerman, 2024-08-07:
Clearing the platform is still working for me; what hardware are you on where you got this error?

```
> sudo tpm2_clear
WARNING:esys:src/tss2-esys/api/Esys_Clear.c:291:Esys_Clear_Finish() Received TPM Error
ERROR:esys:src/tss2-esys/api/Esys_Clear.c:97:Esys_Clear() Esys Finish ErrorCode (0x0000098e)
ERROR: Esys_Clear(0x98E) - tpm:session(1):the authorization HMAC check failed and DA counter incremented
ERROR: Unable to run tpm2_clear
> sudo tpm2_clear -c platform
```

2004seraph, 2025-10-04:
Using @duplexsystem solution fixed it for me too. @TobiPeterG I had completely cleared the installed keys in my BIOS and disabled secure boot when I ran the command, try that.

2004seraph, 2025-10-04:
Actually, after having to repeat the steps here, my TPM seems to be inaccessibly bricked.

Ive tried clearing and restoring the keys in the BIOS, enabling secure boot, disabling secure boot, I can't for the life of me reset the TPM anymore.

For some reason `sudo tpm2_clear -c platform` outputs:

I have no idea why the disable TPM clearing flag is now on, but none of the commands here help me to undo it.