system76/firmware-open #506: darp8 doesn't enter SLP_S0

DrymarchonShaun, 2024-01-11:
- Model: darp8 
- BIOS version: 2024-01-10_6c402c3 
- EC version: 2024-01-10_6c402c3 
- OS: NixOS 23.11
- Kernel: 6.1.69 and 6.6.8

intel's s0ix-selftest-tool is saying the device isn't reaching the deepest state of s0ix, log for that is [here](https://github.com/system76/firmware-open/files/13895904/20240110-16-39-s0ix-output.log).

Manually checking the actual files I'm assuming that script checks I'm finding the same, not reaching slp_s0 -
```
$ sudo cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec
0
```
```
$ cat /sys/devices/system/cpu/cpuidle/low_power_idle_system_residency_us
0
```
however, the CPU is hitting C10 - 
```
$ cat /sys/devices/system/cpu/cpuidle/low_power_idle_cpu_residency_us
1115299596
```
```
$ sudo cat /sys/kernel/debug/pmc_core/package_cstate_show                      
Package C2 : 434296807
Package C3 : 300541816
Package C6 : 61166
Package C7 : 0
Package C8 : 252830
Package C9 : 0
Package C10 : 105541521
```
### Steps to reproduce

Suspend the system
Resume 
Run `sudo cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec`

### Expected behavior

`sudo cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec` should return  greater than 0

### Actual behavior

`sudo cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec` returns 0

crawfxrd, 2024-01-18:
I don't know what's required for `SLP_S0#`, but I expect a part of it is missing or incorrect RTD3 configs.

DrymarchonShaun, 2024-01-20:
> I don't know what's required for `SLP_S0#`, but I expect a part of it is missing or incorrect RTD3 configs.

I assume that would be what's causing the `Pcieport is not in D3cold:` parts of the selftest-tool's output?
```
Checking PCI Devices D3 States:
[  309.689726] nvme 0000:2f:00.0: PCI PM: Suspend power state: D0
[  309.689730] nvme 0000:2f:00.0: PCI PM: Skipped
[  309.691814] i801_smbus 0000:00:1f.4: PCI PM: Suspend power state: D0
[  309.691817] i801_smbus 0000:00:1f.4: PCI PM: Skipped
[  309.693959] pcieport 0000:00:1d.0: PCI PM: Suspend power state: D0
[  309.693962] pcieport 0000:00:1d.0: PCI PM: Skipped
[  309.695756] snd_hda_intel 0000:00:1f.3: PCI PM: Suspend power state: D3hot
[  309.695762] i915 0000:00:02.0: PCI PM: Suspend power state: D3hot
[  309.696360] xhci_hcd 0000:00:0d.0: PCI PM: Suspend power state: D3hot
[  309.702096] r8169 0000:2e:00.0: PCI PM: Suspend power state: D3hot
[  309.705955] sdhci-pci 0000:2d:00.0: PCI PM: Suspend power state: D3hot
[  309.706697] nvme 0000:01:00.0: PCI PM: Suspend power state: D3hot
[  309.706773] mei_me 0000:00:16.0: PCI PM: Suspend power state: D3hot
[  309.706825] pcieport 0000:00:1c.0: PCI PM: Suspend power state: D0
[  309.706827] pcieport 0000:00:1c.0: PCI PM: Skipped
[  309.707008] intel-lpss 0000:00:15.0: PCI PM: Suspend power state: D3hot
[  309.707208] xhci_hcd 0000:00:14.0: PCI PM: Suspend power state: D3hot
[  309.707892] iwlwifi 0000:00:14.3: PCI PM: Suspend power state: D3hot
[  309.711623] thunderbolt 0000:00:0d.2: PCI PM: Suspend power state: D3hot
[  309.714532] pcieport 0000:00:1c.7: PCI PM: Suspend power state: D3hot
[  309.740534] pcieport 0000:00:06.0: PCI PM: Suspend power state: D3cold

Checking PCI Devices tree diagram:
-[0000:00]-+-00.0  Intel Corporation Device 4621
           +-02.0  Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics]
           +-06.0-[01]----00.0  Sandisk Corp SanDisk Ultra 3D / WD Blue SN550 NVMe SSD
           +-07.0-[02-2c]--
           +-0a.0  Intel Corporation Platform Monitoring Technology
           +-0d.0  Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller
           +-0d.2  Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0
           +-14.0  Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller
           +-14.2  Intel Corporation Alder Lake PCH Shared SRAM
           +-14.3  Intel Corporation Alder Lake-P PCH CNVi WiFi
           +-15.0  Intel Corporation Alder Lake PCH Serial IO I2C Controller #0
           +-15.1  Intel Corporation Alder Lake PCH Serial IO I2C Controller #1
           +-16.0  Intel Corporation Alder Lake PCH HECI Controller
           +-1c.0-[2d]----00.0  O2 Micro, Inc. SD/MMC Card Reader Controller
           +-1c.7-[2e]----00.0  Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
           +-1d.0-[2f]----00.0  Sandisk Corp SanDisk Ultra 3D / WD Blue SN570 NVMe SSD (DRAM-less)
           +-1f.0  Intel Corporation Alder Lake PCH eSPI Controller
           +-1f.3  Intel Corporation Alder Lake PCH-P High Definition Audio Controller
           +-1f.4  Intel Corporation Alder Lake PCH-P SMBus Host Controller
           \-1f.5  Intel Corporation Alder Lake-P PCH SPI Controller

The pcieport 0000:00:1d.0 ASPM enable status:
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk+

Pcieport is not in D3cold：          
0000:00:1d.0

The pcieport 0000:00:1c.0 ASPM enable status:
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes, Disabled- CommClk-

Pcieport is not in D3cold：          
0000:00:1c.0

Pcieport is not in D3cold:     
0000:00:1c.7

Available bridge device: 0000:00:06.0 0000:00:07.0 0000:00:1c.0 0000:00:1c.7 0000:00:1d.0

```

peterpeterp, 2024-06-21:
I think I have the same issue on my Thinkpad T14 Gen4 with manjaro
Did you fix it?

DrymarchonShaun, 2024-06-21:
> I think I have the same issue on my Thinkpad T14 Gen4 with manjaro
> Did you fix it?

I haven't checked to see if it's still an issue recently but as far as I know it hasn't been fixed.

danielstuart14, 2024-08-12:
Exact same issue on a Lenovo V14 (i5 12th Gen). For me this screams a Kernel bug, instead of a EC problem.
s0ixSelftestTool also states that the NVME ssd / controller is the culprit, but on my case it is a Samsung PM9B1.