system76/firmware-open #509: darp8 keyboard debounce too lax resulting in multiple chharacters

uSpike, 2024-01-19:
- Model: darp8
- BIOS version: 2024-01-10_6c402c3
- EC version: 2024-01-10_6c402c3
- OS: ubuntu 20.04
- Kernel: 6.6.6-76060606-generic

Keys are sommetimes repeated when typing normally.  Typing in this issue is intentionally noot coorrected.

### Steps to reproduce

Type a bunch of text and eventually kkeys will be repeated.

### Expected behavior

OOne press of the key will result in one character typed.

### Actual behavior

Sometimes two characters are typed.

### Additional info

Probably related to #471

uSpike, 2024-01-19:
@leviport mentioned https://github.com/system76/ec/pull/405#pullrequestreview-1616334945 that they could nnoot tell the difference between two different moodels?  Is it possible that this is another Clevo change since yoour engineering samples?  Also it seems that some letters are more susceptible to repeats such as "o"

uSpike, 2024-01-19:
Indeed it seems that "o" and adjacent keys are repeated most often:

```
The brown fox jumps over the lazy dog
The brown fox jumpps over the lazy dog
The brroown foox jumps over the lazy dog
The brown foox jumpps over the lazy dog
The brown foox jumps over the lazy doog
The brown fox jumps over the lazy dog
The brown foox jumps over the lazy dog
The brown fox jumps over the lazy dog
The broown fox jumps over the lazy dog
The broown fox jumps over the lazy dog
The broown foox jumps over the lazy doog
The brown foox jumps over the lazy dogg
```

leviport, 2024-01-19:
I doubt it's a hardware difference. Keyboard debounce is a highly subjective setting. When we had it set to 5ms, I didn't notice any repeats. Changing it to 10ms fixed it for most. Typing styles vary widely across different individuals.

Are you comfortable building and flashing EC on your machine? I'd be curious whether changing it back to 15ms is sufficient for you. Since 5ms doesn't cause repeated keystrokes for my typing style, the testing I can do myself is quite limited.

uSpike, 2024-01-19:
15ms is better,, but i'll try 20

```diff
diff --git a/src/board/system76/common/kbscan.c b/src/board/system76/common/kbscan.c
index 4d3b650..a5ef0a7 100644
--- a/src/board/system76/common/kbscan.c
+++ b/src/board/system76/common/kbscan.c
@@ -21,7 +21,7 @@
 #endif // KM_NKEY
 
 // Debounce time in milliseconds
-#define DEBOUNCE_DELAY 10
+#define DEBOUNCE_DELAY 15
 
 // Deselect all columns for reading
 #define KBSCAN_MATRIX_NONE 0xFF
```

```
The quick brown fox jumps over the lazy doog
The quick brown fox jumps over the lazy dog
The quick brown fox jumps over the lazy doog
The quick brown fox jumps over the lazy dog
The quick brown fox jumpps over the lazy dog
The quick brown fox jumps over the lazy doog
The quick brown fox jumps over the lazy doog
The quick brown fox jumps over the lazy dog
The quick brown fox jumps over the lazy dog
```

uSpike, 2024-01-19:
20ms seems to do the trick, I can type over 10 lines with no repeats.

Could y'all be convinced to up the value to 20ms?

ShabazzKing, 2024-04-01:
I hhave the same problem with darp7, since I have bought this machine. I thought it's a haardware issue, so I didn't open the issue. Please System76, fix this problem by setting debounce even to 30 ms.

leviport, 2024-04-01:
@ShabazzKing have you tried 20ms? We don't want to go too high, or quick presses will get ignored.

ShabazzKing, 2024-04-01:
> @ShabazzKing have you tried 20ms? We don't want to go too high, or quick presses will get ignored.

No, because I'm not able to compile firmware byy myself, I don't have enough expperience. I can only compile userspace programs, like VLC media player or Strawberry. But of course, I suspect, that 20 ms will do the work.

uSpike, 2024-04-03:
I've been using 20ms for some time now and while I do get repeated characters, they are infrequent enough as to not be annoying.  Maybe a couple per day.

I find it incredible that anyone could use this hardware with the settings at 10ms.  I understand there are differences in typing style but even if I'm careful and slow when typing I get a lot of repeated characters.

aescgar, 2024-05-03:
> If there's no perfect default that we can settle on, we may have to make this an end-user configurable setting.

This is absolutely the right answer, if doable. **Thehre** can be a sane default while still allowing **useres** to set their own tweaks. As you can see, I would also benefit from having this as a customizable setting.

uSpike, 2024-05-04:
I agree that making it user configurable is the best solution.  Use case: I needed to use my wife's laptop (same model) and immediately I had lots of repeated chars.  So many that it was difficult to do what I needed which was simple terminal commands to upgrade packages and things.  No way am I going to flash firmware to use her machine for a few minutes!

ShabazzKing, 2024-09-25:
Is there any pprogress in this issue?