system76/firmware-open #511: `lemp12` usb-c port detecting devices at usb 2.0 speeds

thomas-zimmerman, 2024-01-27:
- Model: lemp12
- BIOS version: 2023-09-08_42bf7a6
- EC version: 2023-09-08_42bf7a6
- OS: Pop!OS 22.04
- Kernel: 6.5.6 (and 6.6.10)

This was found with a Gigabyte USB Ethernet adapters that all are limited to ~300MB speeds when attached to the USB-C port but get full speeds when moved to the USB-A ports.

### Additional info

Looking at the kernel logs, devices attached to the USB-A port are detected as "SuperSpeed" devices, while all of the devices that I attached via the USB-C port are getting "high-speed" settings from the kernel.

crawfxrd, 2024-06-11:
Is this reproducible with storage devices?

Is it limited to lemp12 or do other ADL/RPL systems have the issue?

leviport, 2024-06-12:
Yep, benchmarking my USB-C SSD that I use for Ventoy with Gnome Disks shows me speeds around 40MB/s.

Benchmarking the same drive on both USB-C ports on darp8 gives me ~1GB/s

leviport, 2024-06-12:
After flashing https://github.com/system76/coreboot/pull/227, I'm seeing benchmark speeds on the USB-C port hit ~1GB/s as expected

leviport, 2024-07-31:
The firmware update fixing this has passed testing and is currently being released.

closed 2024-07-31 by leviport