system76/firmware-open #538: Consider using the me_cleaner fork that works with ME  12, 14, 15 and 16 for true deblobbing and stripping of the ME

ilikenwf, 2024-04-01:
https://github.com/corna/me_cleaner/pull/384

I have been using this plus the HAP bit for months now without issue, I know it would require testing with basically all System76 machines ME 12+, but would it not be worth it and another selling point? 

I know that you're using the Alt disable bit, but this is even more...forceful/assured?

crawfxrd, 2024-04-01:
Fuck Intel for what they've done and are doing, but I'm not going to do this based off of just some PR for an unmaintained project that we've never actually vetted or used.

So the process for CSME is:

- Spend 5 minutes logging in to the stupid fucking Intel site, requiring 2 fucking security passes (since they switched to Microsoft's shit).
- Spend 10 minutes trying using the shit search implementation to make sure I find the actual latest Best Known Configuration (BKC) for the specific model (because there's now, like, at least, 3 versions for each SoC generation: U/P/H/HX/S).
- Regenerate the `fit.xml` file for the latest BKC, with:
  - Locking the descriptor disabled
  - Boot Guard disabled
  - PTT (CSME TPM) disabled (all our laptops have Infineon chips)

coreboot remains responsible for disabling the CSME during early boot via the HECI command.

But from what I've seen (between coreboot code and actual behavior), that may not even work correctly anymore...

closed 2024-04-01 by crawfxrd

ilikenwf, 2024-04-01:
I love the fire in your belly, haha!

That said, despite having enabled HECI, I guess it's not truly defined in the devicetree or something as I can't use intelmetool to confirm that ME is truly neutered, but that said, me_cleaner is indeed a vetted project - the 12-16 stripping is done by another guy, though...but the actual me_cleaner repo is within the utils dir of coreboot and part of the build system when enabled for older ME versions.

In the case of ME 16, no deblobbing actually happens but the HAP bit is set. I figure that having it in addition to the HECI disable should be a double proof method of restricting the potential risks, as well as using a non-Intel wlan card.

The fun part is, while you're under contract, all those files and tools you mention largely "fell off the truck" on a certain forum. I've played with them before.

crawfxrd, 2024-04-01:
The HECI device gets hidden at the end of coreboot's run before jumping to the payload, so it won't be visible to the OS.

ilikenwf, 2024-04-01:
I added a line to disable that, which didn't work - even commenting out the function that hides it's body didn't work, either.

````
CONFIG_DISABLE_HECI1_AT_PRE_BOOT=n
````