system76/firmware-open #540: `serw13` suspend not working in Windows 11

thomas-zimmerman, 2024-04-05:
- Model: serw13
- BIOS version: 2023-09-19_16ef69c
- EC version: 2023-09-19_16ef69c
- OS: Windows 11 23H2
- Kernel: Windows 11 Enterprise

Suspending the system, screen goes black but not able to wake the system.

### Steps to reproduce

Suspend the system.  We do see the power LED transition to a slow blink. Any key press to wake the system, switches to a sold green with disk acitvity blinking but the screen does not wake up.

### Expected behavior

Suspend and resume works.

vittvolt, 2024-04-08:
Same on Ubuntu 22.04 Linux kernel 6.8.0 for my Thelio. Device enters suspension but will never come out of it.

crawfxrd, 2024-06-11:
Do any other RPL system have this issue?

thomas-zimmerman, 2024-06-14:
Checking customer issues in Zendesk, I'm not seeing this reported by customers on the `oryp12` or `bonw15`

leviport, 2024-06-14:
I think oryp12 might also be affected. It sticks on a black screen when resuming for a good ~20 seconds, then BSoD's and reboots.

crawfxrd, 2024-06-17:
2024-06-13_b652dda is staged for serw13 to test if updating to coreboot 24.02 addressed the issue.

leviport, 2024-06-17:
I tested `2024-06-13_b652dda` on serw13, but it seems the problem persists. 

I should have been a little more specific when comparing with oryp12, however. The serw13 never shows a BSoD, but it does reboot after about 20 seconds or so of sitting on a blank black screen.

crawfxrd, 2024-06-17:
Does the BSOD on oryp12 provide any details about the cause?

leviport, 2024-06-17:
It only sticks around for a second or two, but I grabbed a picture:
[image]