system76/firmware-open #545: Hibernate inconsistent memory map error when charging with barrel port

mkapalka, 2024-05-03:
- Model: lemp10
- BIOS version: 2023-10-13_5d997d1
- EC version: 2023-10-13_5d997d1
- OS: Pop!_OS 22.04
- Kernel: 6.8.0-76060800daily20240311-generic

When laptop is charging using the barrel connector, resume from hibernate fails with error "Hibernate inconsistent memory map". It works fine when charging with USB-C or when on running on battery.

### Steps to reproduce

Connect barrel connector charger, hibernate, power on the laptop again.

### Expected behavior

Should see back the opened windows.

### Actual behavior

Boots into initial login screen, dmesg shows the hibernate error above.

### Additional info

It seems that charging using the barrel connector changes the memory mapping of the firmware after resume from hibernate:

Before hibernate:
```
BIOS-e820: [mem 0x0000000100000000-0x000000047e358fff
BIOS-e820: [mem 0x000000047e359000-0x000000047e359fff
BIOS-e820: [mem 0x000000047e35a000-0x000000047e35bfff
BIOS-e820: [mem 0x000000047e35c000-0x000000047e450fff
BIOS-e820: [mem 0x000000047e451000-0x000000047e451fff
BIOS-e820: [mem 0x000000047e452000-0x000000047f416fff
```

After failed resume:
```
BIOS-e820: [mem 0x0000000100000000-0x000000047e35bfff
BIOS-e820: [mem 0x000000047e35c000-0x000000047e35cfff
BIOS-e820: [mem 0x000000047e35d000-0x000000047e35efff
BIOS-e820: [mem 0x000000047e35f000-0x000000047e453fff
BIOS-e820: [mem 0x000000047e454000-0x000000047e454fff
BIOS-e820: [mem 0x000000047e455000-0x000000047f416fff
```
When charger is disconnected, I get consistently the latter mapping.

virtadpt, 2024-05-23:
I'm running into this as well (lemp13), only it happens whether or not the unit is plugged in (USB-C is the only supported charging method) or not.  It's been intermittant (it fails more than it succeeds).  From the kernel message buffer:

```
PM: Loading and decompressing image data (3573928 pages)...
Hibernate inconsistent memory map detected!
PM: hibernation: Image mismatch: architecture specific data
PM: hibernation: Read 14295712 kbytes in 0.01 seconds (1429571.20 MB/s)
PM: Error -1 resuming
PM: hibernation: Failed to load image, recovering.
PM: hibernation: Basic memory bitmaps freed
OOM killer enabled.
Restarting tasks ... done.
video LNXVIDEO:00: Restoring backlight state
PM: hibernation: resume failed (-1)
```

nextsux, 2024-08-12:
me too on lemp13 :+1:

gldickens3, 2026-08-31:
Reproduced on darp10 with newer firmware
Model: Darter Pro (darp10)
BIOS: 2025-04-03_04c4e20
OS: Kubuntu (kernel 7.0.0-30-generic)
Bootloader: GRUB2
I can confirm this bug is still present on the 2025-04-03 firmware. Resume from hibernate fails intermittently on cold boot with the exact same signature:
plain
PM: Image signature found, resuming
PM: hibernation: resume from hibernation
...
Hibernate inconsistent memory map detected!
PM: hibernation: Image mismatch: architecture specific data
PM: Error -1 resuming
PM: hibernation: Failed to load image, recovering.
PM: hibernation: resume failed (-1)
Failure rate is roughly 5–10 % of cold-boot resume attempts. The hibernation write itself always succeeds; the failure is strictly at resume time when the BIOS reports a shifted e820 map compared to the write boot.
Since this reproduces on a newer firmware revision than 2024-09-11_351d80d, the issue does not appear to have been addressed in the 2025-04-03 update. Happy to provide dmesg dumps or any other diagnostics if helpful.

marbris, 2026-09-18:
Confirming this still occurs on newer hardware and firmware — and that it is **not limited to the barrel/DC port**.

**System:** Lemur Pro (`lemp12`)
**Firmware:** `2025-07-24_c242738` (coreboot 25.05, EC same) — reproduces on the latest System76 Open Firmware as of this writing.
**OS:** Arch Linux, kernel `7.2.6-arch2-1`, udev-based initramfs with the `resume` hook, `resume=UUID=…` on the kernel command line, dedicated 16 GiB swap logical volume.

**Symptom:** identical to the original report. Resume proceeds until the kernel compares the e820 map, then aborts:

```
PM: Image signature found, resuming
PM: Loading and decompressing image data (1431972 pages)...
Hibernate inconsistent memory map detected!
PM: hibernation: Image mismatch: architecture specific data
PM: Error -1 resuming
PM: hibernation: resume failed (-1)
```

**Charge-state correlation:** hibernation resume **fails while charging** and **succeeds on battery** (unplug before hibernating and resume works reliably). On this lemp12 it fails with **both the barrel/DC jack and USB-C charging** — the most recent failure above was captured while charging over USB-C. So whatever the firmware changes in the memory map when charging is not specific to the barrel port.

**e820 evidence (USB-C charging):** diffing `BIOS-e820` between the boot that hibernated and the boot that tried to resume shows the ACPI / device-reserved regions near the top of RAM shifting by ~28 KB:

```
hibernate boot:  [mem 0x…47e816000-0x…47e816fff] device reserved
                 [mem 0x…47e817000-0x…47e821fff] ACPI data
resume boot:     [mem 0x…47e81d000-0x…47e81dfff] device reserved
                 [mem 0x…47e81e000-0x…47e828fff] ACPI data
```

That shift is what trips `hibernation_e820_mismatch()`, so the kernel refuses to restore. It's consistent with the theory that charge state changes the firmware-reported memory map between hibernate and resume. Happy to provide full dmesg / e820 dumps if useful.