system76/firmware-open #566: [darp8] Setting charge end threshold silently fails after BIOS update

pkowalczyk-efi, 2024-07-11:
- Model: darp8
- BIOS version: 2024-01-10_6c402c3
- EC version: (see BIOS version)
- OS: Arch Linux (freshly updated)
- Kernel: 6.9.8-arch1-1

In the past I was able to set charge end threshold via sysfs. Today I have updated my darp8's BIOS from some 2022 (probably 2022-11-21) to 2024-01-10 and now setting the threshold has no effect.

### Steps to reproduce
Make sure your battery is charged over 20%.
```
echo 20 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

### Expected behavior
Setting the threshold should:
1. stop charging after meeting the threshold (or instantly if it's already met)
1. reading the threshold value with `cat /sys/class/power_supply/BAT0/charge_control_end_threshold` should return the  threshold previously set

### Actual behavior
```
$ cat /sys/class/power_supply/BAT0/status 
Charging
$ cat /sys/class/power_supply/BAT0/capacity
75
$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
100
$ echo 70 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold 
70
$ cat /sys/class/power_supply/BAT0/status 
Charging
$ cat /sys/class/power_supply/BAT0/charge_control_end_threshold
100
```

### Additional info
Stock kernel, no extra system76-* packages from AUR installed.

pkowalczyk-efi, 2024-07-11:
OK, I just noticed that `charge_control_start_threshold` is 90 by default. When I set it to 20, then setting `charge_control_end_threshold` works, as mentioned in "expected behaviour"...

crawfxrd, 2024-07-11:
2023-09-08_42bf7a6 includes the change:

- Changed battery charge start threshold to 90%

When configuring charge thresholds, the start threshold should be updated first. The end threshold must be greater than the start threshold.

closed 2024-07-11 by crawfxrd

pkowalczyk-efi, 2024-07-12:
OK, seems logical, however inconsistent.
I have never used `charge_control_start_threshold`. As far as I understand it tells the charging circuitry on what (dis)charge level to start charging. In the past I could arbitrarily set `charge_control_end_threshold` and charging would start as soon as the charger was plugged in. With current BIOS that's no longer the case, which is not ideal.

In other words - to restore previous behaviour, a special value of `0` must be set on `charge_control_start_threshold`.  Will propose a change to https://github.com/system76/docs/edit/master/content/laptop-battery-thresholds.md, elaborating on this edge case.