system76/firmware-open #573: coreboot.org site not connecting when running openfirmware dependecies script

robotparty, 2024-08-10:
- Model: gaze17-3060-b
- BIOS version: 2022-06-07_090f9e0
- EC version: 
- OS: Pop!_OS 22.04 LTS \n \l
- Kernel: 6.9.3-76060903-generic

Using the https://github.com/system76/firmware-open/blob/master/scripts/install-deps.sh script to install dependecies for firmware update but it can't complete. Something wrong with a code source website?

### Steps to reproduce

run installdep.sh script

### Expected behavior

open firmware dependencies should be installed

### Actual behavior
script in retry loop
```
Cloning into '/home/chris/git/firmware-open/coreboot/3rdparty/ffs'...
fatal: unable to access 'https://review.coreboot.org/ffs.git/': Failed to connect to review.coreboot.org port 443 after 135134 ms: Connection timed out
fatal: clone of 'https://review.coreboot.org/ffs.git' into submodule path '/home/chris/git/firmware-open/coreboot/3rdparty/ffs' failed
Failed to clone '3rdparty/ffs'. Retry scheduled
Cloning into '/home/chris/git/firmware-open/coreboot/3rdparty/fsp'...
fatal: unable to access 'https://review.coreboot.org/fsp.git/': Failed to connect to review.coreboot.org port 443 after 135103 ms: Connection timed out
fatal: clone of 'https://review.coreboot.org/fsp.git' into submodule path '/home/chris/git/firmware-open/coreboot/3rdparty/fsp' failed
Failed to clone '3rdparty/fsp'. Retry scheduled
Cloning into '/home/chris/git/firmware-open/coreboot/3rdparty/intel-microcode'...
```

### Additional info

crawfxrd, 2024-08-12:
Not a firmware-open issue. gerrit was likely unavailable, which happens sometimes.

2023-09-08_42bf7a6 is available for gaze17-3060-b through Systm76 firmware manager.

closed 2024-08-12 by crawfxrd

robotparty, 2024-08-12:
Seems like it IS an openfirmware issue because without Gerritt, the open firmware tooling is useless. 
The reliability of the infrastructure & tool chain IS the tool at some point. 

Can we use the coreboot github mirror somehow when it's down?