system76/firmware-open #575: building galp5 firmware from source results in intermittant boot issues

loftlifter31, 2024-08-27:
- Model: galp5
- BIOS version: built from source
- EC version: built from source
- OS: PopOS 21
- Kernel: 5.16
Built from source using both master branch and edk2-stable branch. Running flash.sh reboots into system-update efi but then says no updates found. Imaging usb.img onto usb drive and booting from there does successfully update the firmware and ec, but rebooting into the new firmware takes me to a black screen. After rebooting several times it does finally boot into the OS using the new firmware, but on next reboot black screen again. Successfully boots 1 out of 3 or 4 tries. Have also tried using external firmware programmer to flash BIOS with same results, either using rom built from master or edk2-stable. Re-flashed original ROM and there are no boot issues, even using the new EC.

### Steps to reproduce

1. Build firmware.rom and ec.rom from source
2. Update galp5 firmware with built from source rom
3. Reboot 

### Expected behavior

Boot into OS

### Actual behavior

Intermittant boot failures

### Additional info

Even on successful boot, the source firmware does not present a boot options menu when ESC is pressed. Instead there is an error message which disappears too fast for me to read.
[system76_build.log](https://github.com/user-attachments/files/16767975/system76_build.log)
[dmesg.new.rom.txt](https://github.com/user-attachments/files/16768004/dmesg.new.rom.txt)
[dmesg.original.rom.txt](https://github.com/user-attachments/files/16768005/dmesg.original.rom.txt)

crawfxrd, 2024-08-27:
There is no `edk2-stable` branch. The only branch you should be using is `master`.

loftlifter31, 2024-08-28:
[image]

crawfxrd, 2024-08-28:
Yes, there is a branch called `edk2-stable202405`, not `edk2-stable`. And it's a development branch; you should not be using it.

loftlifter31, 2024-08-28:
As I mentioned, I built from both edk and master branch and both demonstrate the same issue.

crawfxrd, 2024-08-28:
You did not provide the commit you used. You have a modified local repo, because the version in the build log is "1.0" and not the commit. If you are using the master branch, you are on an old commit, because the Rust toolchain is using nightly-2023-09-07.

loftlifter31, 2024-08-29:
The commit to master I am workng with is 4b32a3e9f5ebeabfa0549bdf2eabb0fd3899f893

crawfxrd, 2024-08-29:
Use 5002c9ad95bd or newer. The VBTs for TGL had to be regenerated because parsing stopped working after an FSP update.

loftlifter31, 2024-08-29:
I was using commit 6a6f7f8db01a366c98246f5847f58a266d0f7f10 of coreboot. So you are saying the vbt.rom included with this commit doesn't work with the parsers included in the open-firmware repo for the commit of that I was using? If so, that seems to be bad luck on my part as the 2 commits are only a few days apart. https://github.com/system76/coreboot/compare/6a6f7f8db01a366c98246f5847f58a266d0f7f10...792996fc8cd673f24335ae52e5b2066cccba2f77#diff-441eb6832cbe04acdf2a92203b3d4587a983102af3f391fc951782ac7d76593e

loftlifter31, 2024-08-31:
@crawfxrd I can confirm the vbt.rom was the culprit here. For future reference, how do I know which version of vbt to use with which version of coreboot? I am experimenting with other coreboot payloads and running into similar problems.

crawfxrd, 2024-09-01:
Normally, you just use whatever is in the coreboot repo. In this case, coreboot/coreboot@4c7e97b26a34 to coreboot/coreboot@5d1494adda44 will have broken graphics on our TGL boards.

Ref: #525

loftlifter31, 2024-09-03:
@crawfxrd Are you saying that those versions of coreboot you listed won't work on system76 tgl boards at all or a different vbt must be used?

closed 2024-09-09 by loftlifter31

tlaurion, 2024-09-09:
From https://github.com/linuxboot/heads/issues/1765#issuecomment-2338646515
> I got it working. Just needed to update vbt.