system76/firmware-open #577: EC not requiring physical proof of presence before updating BIOS

loftlifter31, 2024-09-09:
Flashed ec built from master branch onto galp5 with CONFIG_SECURITY=1 set in models/galp5/ec.config file. Flashing new firmware does not require me to enter proof of physical presence code which I thought firmware security was supposed to require. Is there some additional configuration required?

loftlifter31, 2024-09-11:
This is still not functioning as expected. As a test I reflashed the stock firmware that shipped with the machine. Flashed the usb.img for a build completed last week onto a usb drive and booted from that usb. The firmware updated to one which should include support for EC security. I then built a new firmware using the same source code but since it has a newer date the resulting usb image detected the firmware as an update and applied it. Both firmware.rom and ec.rom got updated. At no point was any prompt requiring me to input anything seen. The commit of ec I am using is 88c77aa1d322d2cca56038b396b31a96bc42fe59 I believe this should support CONFIG_SECURITY

crawfxrd, 2024-09-11:
You need to give the commit of firmware-open you are using, and the commit of any submodule if you have modified them.

loftlifter31, 2024-09-12:
firmware open is 4b32a3e9f5ebeabfa0549bdf2eabb0fd3899f893
Is the code to require the proof of presence on update located in the firmware.rom or ec.rom?

crawfxrd, 2024-09-13:
The feature requires the functionality be added/enabled in:

- coreboot
- ec
- edk2
- firmware-setup
- firmware-update

loftlifter31, 2024-09-13:
@crawfxrd How is the functionality enabled in these other submodules? Is this something that needs to change in the source code or something I can enable in a config file?

crawfxrd, 2024-09-13:
If you are using 4b32a3e9f5eb without any modifications to submodules, then the only thing needed is adding `CONFIG_SECURITY=y` in `models/galp5/ec.config`.

loftlifter31, 2024-09-17:
That's just what I did, hence the surprise that it didn't work.

crawfxrd, 2024-09-17:
It must be `y`, not `1`. Can you confirm you have the setting correct.

loftlifter31, 2024-09-18:
Oh, I overlooked that details. I'll try it again with the y instead of 1.

closed 2024-09-27 by crawfxrd