system76/firmware-open #592: bonw15: G-Sync Not Enabled?

ilikenwf, 2024-12-30:
- Model: bonw15
- BIOS version: current
- EC version: current
- OS: Archlinux
- Kernel: cachy

G-Sync does not appear to be setup anywhere in the gpio.c file as it is with the serw15 and others, but the laptop itself should have the capability - the BOE 144hz LCD supports it, the 40XX gpus support it...it's just something somewhere in firmware does not support it.

I've tried both in hybrid and in dedicated graphics mode without luck in both Linux and Windows...it makes no sense.

I'm at a loss, I can't figure it out...

Schenker is legally sharing the schematics for this particular model, under a different name, and there is information in that document relevant (see bottom of page) - I can't find any GPIO directly labeled for g-sync detection but there's an nvidia page that discusses somethign to do with vendor bits for that?

https://download.schenker-tech.de/package/schenker-key-17-pro-e23-e24-ske17pe23-ske17pe24/

### Steps to reproduce

Try to use G-sync in Linux and Windows, in hybrid and dgpu only mode. It will fail.

### Expected behavior

### Actual behavior

### Additional info

ilikenwf, 2024-12-31:
Is this something to do with some GPIO needing pulled, or is it something to do with adding VRR support or configuring the dGPU vendor bit? 

The x370snx models on default firmware with the 144hz LCD support it so really shouldn't it work for us too?

ilikenwf, 2025-01-06:
I'd love some input. Schenker is nice enough to share the x370SNx schematics, so here's the nVidia relevant portion:

[image]

**The full schems (scroll to bottom):**

https://download.schenker-tech.de/package/schenker-key-17-pro-e23-e24-ske17pe23-ske17pe24/

ilikenwf, 2025-01-06:
The other GPIOs on the CPU themselves don't document a g-sync enable pin explicitly like some of the other models but that doesn't necessarily mean that one of the "unused" or non defined ones might be setup for this and just not noted in the schematic?

ilikenwf, 2025-01-06:
@crawfxrd I'm sorry to tag you at all, but I would love it if you or someone else familiar with all this might know a path I could investigate?

ilikenwf, 2025-01-06:
The serw13 at least has it mentioned for the GPIOs, but even if the hardware all supports it they may be artificially turning it off at factory with a resistor? On both it and bonw15? That or there's something elsewhere...driver, vbios, dsdt that prevents it working?

https://download.schenker-tech.de/temp/Manual_Parts-and-Schematics_XMG-PRO-15_PD50SNx.pdf

ilikenwf, 2025-01-06:
Looking in the OEM UEFI modules, it may be that one of the nVidia DXEs does something to enable it? 

The nvdgpudxe does have a list of ceritifications for g-sync...so there may be some special sauce we lack?

ilikenwf, 2025-01-06:
Unless the dxe/smm modules are specifically queried or used somehow, I'm thinking that it could potentially be something that needs to be in the DSDT as an alternative thought, though the relevant certification strings do exist in the stock firmware's nvdgpudxe file...but they look similar to the old SLI "license" entries that used to be used in DSDT if I'm not mistaken.

ilikenwf, 2025-01-06:
I don't have a DSDT dump from the stock bios, and reached out to a guy to try and get one, but if you happen to have one handy feel free to share.

I'm expecting there should be strings somewhere inside like these:

```
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926315            - &U;A5USCTF)\>$K5A33Q'7P"J:$;RF@QBU\I]YHJPA6U - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935362        - SF '_24_"P'&8"2?O%J+AUNK[7=ESCG\SF45==L[?R4% - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935352        - RE/&\37\#Q$%9#1<L$I(@V?KX42D,DF_PG32<A6)6\4# - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
jMKV
952468287716Genuine NVIDIA Certified GSync Ready Platform for TIUYGGARAIZBHIXGOBJT - L1[+*$%\UBPJS\)W)_-N#6^-=Q^Z5,.%E+_AJ0''D HU - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926326            - XW=OBRRDV@+^<"M7C5=S%5R$L$&=PDBWLW^K_[VL^G(W - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for YURQZSXYVVHTXRJZKYHL - F5Y#\IH7L>GARQ>$0"B$S<*,5#"0VCCVB*ZTY"NUHZ<* - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926668            - QA(0QDA5I=DBW\%/=/Q7FWCC.FCQ:-*<78*&-(F_LT<, - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926306            - &U;A5USCTF)\>$K5A33Q'7P"J:$;RF@QBU\I]Y77/F<! - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for GTFPWLWJOCPJNUWALVTV - PZ;D7#>"6%FN),N3XF"8OTF\!+.0R[W&72>45K"? D5] - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
7FW'	
7FW'	
7FW'	
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926325            - 'V<NCSQGWA(]=#L4@4<P$6S#M%'<QEAVMT]H^X60.G=" - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
9muKV
952468287716Genuine NVIDIA Certified GSync Ready Platform for FBMGNXCNHHCXLYUWRNXP - /(W( JD1\@-"OK/P9$G\*K94WFZ#@JGQD!*!?9!4,L:_ - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for LMRSYZJLMFIDERXITRHA - 6>\$UXWK-V0>TQ88_I#9H))R4$9N= >HR\^RQ4JVT7%# - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for TUPKPQUZWQDEQOFKAQBY - [TC\,=36H7Q^:>[\%PK V>]?C5N_3=2]HUV]+1)6-1/X - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935355        - UH"%!08R$R!X> 4!0'T5GS8N]1?8QAIB]@67??1XN[0S - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935368        - IL&!P=?S(^-\",8%4+P1K_74A=;<UMMFYL:+#3B!5X># - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for cl3926666            - __62OF_7G3Z@Q^#-;-W5@)AE,DA_4/(>5>($+6D 1Q0^ - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935353        - SF '_26,"P'&8"2?O%J+AU>L[7=ESCG\SF45==L[?R4% - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935349        - HK% "=9")_"[#-7"5*W6JPD2^28;VNLA^M9("0C 2_9" - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
Jk.KV
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO NB 4062323     - U_0A:.#X= MC,W46_K/F1_MG$VU5UF[+29#+:5"6-L=] - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
fM"KV
952468287716Genuine NVIDIA Certified GSync Ready Platform for CAQPBZDMUKDXBOETIQPR - O4^V/!*HVO]MPY,X*B(K&3[P>R[]:)S(FTBBM7*$I%MZ - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for MCUWNIHEEXAJQOZDTEYU - 5=['RYVH*W1=UR79\H">K*^V5%>M>!?KS]]SPN0&_;'% - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO NB 4062320     - R\=B9!<%8'@@/*15ZH,E2R@Z#U(2PYX$1< 67M[D]A1+ - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for LJBHSRTAVUAMWYUYLOJX - V@1N=-"S<#LDSV59^L(A6^LF'QT6TE](=8$*;AXF&B?[ - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO NB 4062308     - HB3L?+&[>-NFQT7;@N*C4\>C)SV(ZC^.?:&,=CZ@XL=% - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935366        - WJ$##>8#&\#Z<.6#2)V7EQ56_39:WOK@_B8)!J;T9S;& - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
[... cut ...]
952468287716Genuine NVIDIA Certified GSync Ready Platform for ATFDSCFBNEBDEMTUREHT - ? WP0!%Y/[,[=!1:H>A&T$_^$_[Z0"'959*Y/-BRV]4U - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935382        - UH"%!08R$R!X> 4!0'T5GSHI]1?8QAIB]@67??1XN[0S - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
952468287716Genuine NVIDIA Certified GSync Ready Platform for CJONXPUIMTJGZMOHSBOG - 3;Y9P[R](U7;WT57ZF <E,,/3;<K8#=EQS[Q.3IKS4&  - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)
```

ilikenwf, 2025-01-06:
In addition, if the platform cookie requires modification, it appears to be easily doable:

https://web.archive.org/web/20150704043502/http://gamenab.net/2015/07/03/bios-information-for-g-sync/

crawfxrd, 2025-01-06:
I had a mostly-written out response to this, but apparently closed the tab and can't recover it.

I'm too drunk at this point, but in effect:

I don't know how G-SYNC (or equivalent) _actually_ works at the hardware/firmware-level to properly response to this. I think its mostly controlled by ACPI at this point? But the _notion_ of "mid" from that table is so fucked up to me that I don't know where to begin.

For most features, we typically only have to enable it by GPIO or FSP selection.

In terms of security: I have expressly tried to avoid any kind of responsibility for its implementation. While I had a minor in mathematics, I am ultimately just a code monkey. I am just trying to make these implementation FOSS without Intel or NVIDIA bearing down on us.

ilikenwf, 2025-01-07:
All that said I still think it's the NVdGPUDxe that does whatever it is to turn on GSync, but so far ghidra isn't really showing me what I need to see just yet.

I can share files and info if anyone wants to help.

ilikenwf, 2025-01-08:
@crawfxrd I got further...so for one I'm not sure which model for certain corresponds to my machine but the one below is an educated guess, based on values in hex...

Anyway, in theory this should work if it is correct but it's not - so either the license is incorrect for my model (does anything in the list look familiar or could Sager give you the id for the bonw15?) or, more likely, I think g-sync may not work unless we get control of the mux (https://github.com/Dasharo/dasharo-issues/issues/1024  related) so that it is direct output without the igpu in the way...

```
diff --git a/src/mainboard/system76/rpl/acpi/mainboard.asl b/src/mainboard/system76/rpl/acpi/mainboard.asl
index 63ccf020d5..e0d5de5a86 100644
--- a/src/mainboard/system76/rpl/acpi/mainboard.asl
+++ b/src/mainboard/system76/rpl/acpi/mainboard.asl
@@ -18,6 +18,31 @@ Scope (\_SB) {
 		Scope (PEG2) {
 			#include <drivers/gfx/nvidia/acpi/tigerlake.asl>
 		}
+        Scope (PCI0)
+        {
+            Name (SLIC, Buffer (0x9E)
+            {
+                // You will need to extract the cookie for your specific machine from the nVidia dGPU DXE in your machine's proprietary UEFI bios image
+                "952468287716Genuine NVIDIA Certified GSync Ready Platform for CLEVO 3935368        - IL&!P=?S(^-\\\",8%4+P1K_74A=;<UMMFYL:+#3B!5X># - Copyright 2022 NVIDIA Corporation All Rights Reserved-529314921961(R)"
+            })
+            Device (WMI1)
+            {
+                Name (_HID, "PNP0C14")
+                Name (_UID, "OPT1")
+                Name (_WDG, Buffer (0x14)
+                {
+                    /* 0000 */  0x3C, 0x5C, 0xCB, 0xF6, 0xAE, 0x9C, 0xBD, 0x4E,  // <\.....N
+                    /* 0008 */  0xB5, 0x77, 0x93, 0x1E, 0xA3, 0x2A, 0x2C, 0xC0,  // .w...*,.
+                    /* 0010 */  0x4D, 0x58, 0x01, 0x02                           // MX..
+                })
+                Method (WMMX, 3, NotSerialized)
+                {
+                    // this is in case the driver or vbios calls the func?
+                    CreateDWordField (Arg2, Zero, FUNC)
+                    Return (SLIC)
+                }
+            }
+        }
 #else
 		Scope (PEG1) {
 			#include <drivers/gfx/nvidia/acpi/tigerlake.asl>

```

ilikenwf, 2025-01-08:
Regarding those "cookies" - here's the best reading on it. This guy got legal threats from nVidia though... 

https://web.archive.org/web/20150704043502/http://gamenab.net/2015/07/03/bios-information-for-g-sync/

ilikenwf, 2025-01-08:
...and I can confirm that WMI device node does show up properly under Linux so I'm guessing it's the cookie or the mux

ilikenwf, 2025-01-09:
[image]
[image]

Here's a more organized and concise version for your GitHub reply:

@crawfxrd I got it working as a proof of concept! I included the WMI scope in my DSDT (though I'm not sure if it's required or functional yet). What did work was adding the MUX_CTRL_BIOS GPIO to the bonw15 in the EC and forcing it high at boot.

This approach meant flying blind until my bootloader started, but under Linux, it allowed operation in unverified mode. However, despite having the right hardware and G-Sync enabled, I couldn't confirm much improvement in stuttering using the available Linux testing applications though the indicator does say GSYNC is on. Windows, being windows, isn't happy and probably needs additional ACPI table entries to be happy, as will brightness controls as doing it this way kills your eyes with 100% brightness.

Further ACPI and GPIO work will likely be needed to refine this. My next step is to try getting the Advanced Optimus toggle working with the MUX_CTRL_BIOS GPIO. I suspect the original intent might have been for the FN+LCD key to toggle the GPU MUX.

Happy to share my diffs or push if you really want to look at it or if you'd rather let me keep grinding on it a while before you comment.

ilikenwf, 2025-01-09:
...so that DSDT addition may or may not be working but if it is, the nVidia cookie I'm using to attest GSync certification is incorrect for this model.

crawfxrd, 2025-01-09:
I'm not really following this, but I'm not surprised if it's mostly controlled by ACPI.

I'm not sure if `MUX_CTRL_BIOS` requires any interaction with the PS8461. If it's just the EC GPIO control then it should be simple to add.

ilikenwf, 2025-01-09:
It does appear to be mostly ACPI including the LCD brightness in nvidia mux mode; I believe that what I lack there and in the "certification cookie" department, and some other ACPI bits are what's needed to make windows happy as well...

MUX_CTRL_BIOS is exposed to the EC, some of the models have it defined, some don't, but none of them bring it up or have a facility for flipping it yet...and then we'd also just need to add code somewhere in the process so that coreboot outputs to screen regardless of mux - it's a black screen if you bring that GPIO up until the bootloader starts right now.

ilikenwf, 2025-03-10:
@crawfxrd I may be overthinking this; we may just need the option either in userspace or the bios to change the mux (and if using nvidia mux, force nvidia mode instead of hybrid or compute) and also in this mux, get brightness control working and POST working to the display...

And then in Windows, if we don't enforce nvidia mode, a small helper to toggle like we have in system76-power since it seems Windows doesn't set nvidia-only mode and uses optimus always right now?

ilikenwf, 2025-03-10:
Doing this would add value to existing and future machines as g-sync working properly, at least under Pop, would be a great thing.