system76/firmware-open #600: Feature Request: Power on from restore

ahoneybun, 2025-02-19:
Current firmware behavior for the meer9 is are two of the three options that the non-open firmware offers and it would be beneficial to offer the third for remote boxes that are powered off but need to be power on remotely with devices like a PiKVM.

[image]

thomas-zimmerman, 2025-02-19:
Will be checking behavior again as meeting with Levi, what the customer wants to happen is what current behavior should already be.

thomas-zimmerman, 2025-02-19:
Update -- testing on the released and installed firmware: setting the after power fail setting does not seem to work as expected and turn the system back on. 

From the Intel NUC settings, the expected behavior is "Power On"

leviport, 2025-02-20:
@thomas-zimmerman  I re-flashed the same firmware (just with `sudo system76-firmware-cli schedule` and reboot) on the unit you were experimenting with, set `power_on_after_fail=Enable`, and it now seems to be working as expected.

The other unit I had sitting over by my desk worked as expected without me needing to re-flash it, so I'm not sure what the deal was with the first unit.

leviport, 2025-02-21:
I think I've made a discovery. Pressing the power button while the unit is running seems to make `power_on_after_fail` stop working. This can be a quick press which summons the OS's countdown/confirm dialog, which can be dismissed without even shutting the machine down, or it could be holding the button to force the machine to power off. After the power button has been pressed while the machine is running, reboots, power offs, and AC unplugs/re-plugs are not enough to make `power_on_after_fail` work again, even though nvramtool says that it is still enabled.

I have been able to make the setting work again with a sequence of disabling the setting with nvramtool, powering the machine off, unplugging AC, plugging AC back in, booting up via power button, then re-enabling the setting with nvramtool once again. This doesn't seem like a certain fix, however. I'm still working on a procedure to make the setting start working again with 100% confidence.