system76/firmware-open #610: Consider implementing dGPU mux switching

ilikenwf, 2025-04-01:
Dasharo recently added the support for it; I know that the bonw15 supports it and very likely many others...

https://github.com/Dasharo/dasharo-issues/issues/1024#issuecomment-2768989870

@crawfxrd doing this would allow force enabling g-sync on supported models in the nvidia settings in Linux. Under Windows, it's well...Windows so idk there...

ilikenwf, 2025-05-29:
> Does the bonw15 actually have a mux? I'm curious how you have confirmed this. I just stumbled across this issue when coming here to report another firmware issue but I thought I saw on some other issue in system76 (I think system76-power repo?) where someone from S76 said that none of their models since like 2022 has had a mux switch?

100% yes - I pulled the appropriate pin high by force in the EC and got an nvidia only mux, and in nvidia dgpu mode on top of that managed to enable forced g-sync under linux. Windows would not cooperate...and I had not brightness control so my eyes were fried.

I have the schematics for the x370SNW which is the platform bonw15 is based on...so yes, it has a mux, and anything with a dgpu likely does too.

Scroll to the bottom of the page here for those pdfs: https://download.schenker-tech.de/package/schenker-key-17-pro-e23-e24-ske17pe23-ske17pe24/

I need to push my adjusted EC code sometime, I guess, but it's not a complete implementation with a toggle setup yet or bios integration.

To successfully finish mux switching/choosing we'll need to probably replace the LCD FN hotkey combo to toggle the mux, or, more ideally, with the gpio exposed in the EC code, add a menu option to the bios that lets us both choose the mux, as well as the primary GPU or dGPU or just one of them to use. I believe that some Dasharo machines already have this feature now - amongst others I wish that System76 would backport to their own firmware.

RileyMathews, 2025-05-29:
Oh that's awesome to know! Forgive some ignorant questions here but I'm super curious how to do this on my laptop as well...

By 'pulled the appropriate pin high by force in the EC' do you mean you opened up the laptop and manually flipped a switch? Or is this referring to something you are doing in code to toggle something in bios? I spent a lot of time a few months ago searching around for more information on how to flip a mux switch but the most I could come up with is basically "sorry if your bios doesn't support it your out of luck" and then when I stumbled across the S76 employee implying my model didn't have it I just kind of gave up and resigned myself to running i3/Xorg until this machine dies 🙃.

Any info you can provide for me to do this on my system would be amazing! I am running arch on my system if that changes anything.

ilikenwf, 2025-05-29:
A GPIO is basically an on/off switch. They're used all over the place on modern computing devices, at various levels. So yes, I'm more or less force enabling the mux. The downside of doing it that hacky way is that you'll have a black screen until the OS boots up. What I did was dangerous and for the sake of testing, and because there's not an easy mechanism baked into the EC and bios to toggle things the way we want to set them up, it's not really safe to do using my current code.

If your bios doesn't support it, you are indeed out of luck, but our bios is open source, and our EC is open source too...so it can be done, it just hasn't been (fully) on our machines yet, so they weren't really lying, per se it's just not implemented officially.

I also run arch, but this is something that I'd suggest, until you're at the least somewhat familiar with building and modifying the bios and EC, you steer clear of unless/until we get an official implementation or I find the time to fork and pull request what's needed to do it - because in your case, if something goes wrong, you probably don't have the equipment needed to manually recover from a bad EC or bios flash (though that can be had).

If you're technically minded and can learn the languages involved, though, and get familiar with the EC and coreboot code, we could both throw in on this as time permits us too.

RileyMathews, 2025-05-29:
Hmm I see. Yeah I'm not sure I'm in the best place to tinker with this right now then 😬. I would consider myself pretty technically competent but I have no experience getting that low level in the tech stack. The most I've done in that realm is just updating my BIOS using official tools. And also seeing as how my S76 is my work laptop... yeah not going to do any high risk maneuvers on it. If this was my personal laptop though I would totally be going down this path 😆.

Let me know if there is anything else I can do to help out here! I'll also give @crawfxrd a tag to say there is at least one other user who would absolutely love this functionality in the BIOS!

RileyMathews, 2025-05-29:
@crawfxrd thanks for the clarification on the mux switching models! Is there any chance that anyone from S76 would be picking this up anytime soon or would this have to be an outside contribution?

After looking into what it takes to reflash a broken bios I think I might be up for getting the stuff needed to flash from a spare raspberry pi I have lying around and venturing down this path to see what happens.