system76/firmware-open #617: I cant build firmware for serw13

rezaAdinepour, 2025-05-17:
Hi everybody,

I can't build a coreboot firmware using the `firmware-open` repository.

I cloned the repository and installed all dependencies according to the installation docs.

However, when I run the `./scripts/build.sh serw13` command, the following error appears. How can I fix it?

```bash
    DD         Adding Intel Firmware Descriptor
    IFDTOOL    validate IFD against FMAP
File build/coreboot.pre is 33554432 bytes
No Flash Descriptor found in this image
Segmentation fault (core dumped)
make: *** [src/southbridge/intel/common/firmware/Makefile.mk:51: add_intel_firmware] Error 139
```

rezaAdinepour, 2025-05-17:
I find error here in build log as bellow but why?

```bash
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20241212
Copyright (c) 2000 - 2023 Intel Corporation

coreboot toolchain v2025-01-29_cdcd580bce
    CC         bootblock/soc/intel/meteorlake/bootblock/report_platform.o
    CC         bootblock/soc/intel/meteorlake/bootblock/soc_die.o
Traceback (most recent call last):
  File "/home/serw13/Downloads/system76/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py", line 946, in <module>
    sys.exit(main())
  File "/home/serw13/Downloads/system76/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py", line 935, in main
    SplitFspBin (args.FspBinary, args.OutputDir, args.NameTemplate)
  File "/home/serw13/Downloads/system76/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py", line 797, in SplitFspBin
    fd.ParseFd  ()
  File "/home/serw13/Downloads/system76/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py", line 568, in ParseFd
    raise Exception("ERROR: Invalid FV header !")
Exception: ERROR: Invalid FV header !
    CC         bootblock/soc/intel/meteorlake/espi.o
    CC         bootblock/soc/intel/meteorlake/gpio.o
    CC         bootblock/soc/intel/meteorlake/gspi.o
    CC         bootblock/soc/intel/meteorlake/i2c.o
make: *** [src/drivers/intel/fsp2_0/Makefile.mk:112: build/Fsp_M.fd] Error 1
make: *** Waiting for unfinished jobs....
dsdt.asl     28:    Name (RBUF, ResourceTemplate ()

```

crawfxrd, 2025-06-01:
This is not reproducible. Do you have LFS objects pulled?

bdmstyle, 2025-06-01:
I just donwload a brand new pop is install and follow the https://github.com/system76/firmware-open/blob/master/docs/building.md

After everything run I got 

```
    DD         Adding Intel Firmware Descriptor
    IFDTOOL    validate IFD against FMAP
File build/coreboot.pre is 33554432 bytes
No Flash Descriptor found in this image
Segmentation fault (core dumped)
make: *** [src/southbridge/intel/common/firmware/Makefile.mk:51: add_intel_firmware] Error 139
```

boyangma, 2025-06-07:
Same error ("ERROR: Invalid FV header !") on meer9. The only thing in build/ is UEFIPAYLOAD.fd.

bdmstyle, 2025-07-03:
@crawfxrd The `lfs pull` command resolved the build issue. I suggest adding this step to the build documentation.

See also: [Pull Request #629](https://github.com/system76/firmware-open/pull/629)

---

## Full Command (single line)

```bash
git clone --recurse-submodules https://github.com/system76/firmware-open.git \
&& cd firmware-open \
&& git lfs fetch --all \
&& git lfs pull \
&& git submodule foreach --recursive 'git lfs fetch --all && git lfs pull' \
&& git submodule update --init --checkout --recursive
```

## Install dependencies

```bash
./scripts/install-deps.sh
./scripts/install-rust.sh
./scripts/coreboot-sdk.sh
./ec/scripts/deps.sh
```

## Build

```bash
./scripts/build.sh serw13
```

closed 2025-10-23 by crawfxrd