system76/firmware-open #618: Help with BIOS Issue on Serw13 Laptop

rezaAdinepour, 2025-05-17:
Hello,

I own a Serw13 laptop and am facing an issue with it, so I need your help.

I attempted to modify the BIOS and accidentally flashed an unstable BIOS version onto it. Now, the laptop does not boot. After powering it on, it directly enters the GNU GRUB screen.

Is there any way to reflash the original BIOS without having to physically desolder the BIOS chip from the motherboard?

Thank you in advance for your assistance.

crawfxrd, 2025-05-17:
- [`docs/flashing.md#external-programmer`](https://github.com/system76/firmware-open/blob/4fc740e7897b46b312f0347f2b4c4e604b66cc5f/docs/flashing.md#external-programmer)

closed 2025-05-17 by crawfxrd

rezaAdinepour, 2025-05-18:
How do I do this? I’m not very familiar with GRUB. If possible, could you please help me?

garrettjwilke, 2025-05-19:
you can contact system76 support:
https://support.system76.com/

if you can get into GRUB, that means that you can boot into an OS. download a Pop!_OS iso and the firmware manager should be able to find a bios for you

rezaAdinepour, 2025-05-20:
Thanks a lot! I went to the BIOS recovery directory, executed `old_firmware.elf`, and my system booted without any problems.