system76/firmware-open #625: Compiling firmware for gaze17-3050 fails

chpgyr, 2025-06-08:
- Model:  gaze17-3050
- BIOS version: 2023-09-08_42bf7a6
- EC version: 
- OS:  Fedora Linux 42 (Workstation Edition)
- Kernel: 6.14.9-300.fc42.x86_64

Compilation fails

./scripts/build.sh gaze17-3050

### Expected behavior

Firmware shoud be compiled without any error

### Actual behavior

main.c: In function 'ProcessArgs':
main.c:163:42: error: too many arguments to function 'p->process'; expected 0, have 2
  163 |                                         (*p->process)( *argv, *(argv+1) );
      |         

### Additional info

gcc --version
gcc (GCC) 15.1.1 20250521 (Red Hat 15.1.1-2)

chpgyr, 2025-06-13:
Thanks for the hints. 
I have indeed not installed lfs. After installing lfs, it still doesn't work. The same error 
main.c: In function 'ProcessArgs':
main.c:163:42: error: too many arguments to function 'p->process'; expected 0, have 2
  163 |                                         (*p->process)( *argv, *(argv+1) );
      |                                         ~^~~~~~~~~~~~  ~~~~~
is posted
and there is the information "1.85.0-x86_64-unknown-linux-gnu (overridden by '/home/..../firmware-open/ec/rust-toolchain.toml')" posted too.  Any clue?

crawfxrd, 2025-07-03:
Building coreboot toolchain and host tools is fixed.

Still requires updating edk2 BaseTools.

chpgyr, 2025-07-20:
Hi crawfxrd, I used edk2: Rebase.... However, it still doesn't work.
./scripts/build.sh gaze17-3050
Building '2025-07-08_c4f98eb' for 'gaze17-3050'
make: Entering directory „/home/.../firmware-open/apps/firmware-setup“
cargo build --release
   Compiling autocfg v1.4.0
   Compiling libm v0.2.11
   Compiling orbclient v0.3.48
   Compiling ttf-parser v0.15.2
   Compiling redox_uefi v0.1.13
   Compiling redox_hwio v0.1.6
   Compiling downcast-rs v1.2.1
   Compiling system76_firmware_setup v1.0.0 (/home/cg/Software/firmware-open/apps/firmware-setup)
error[E0463]: can't find crate for `core`
  |
  = note: the `x86_64-unknown-uefi` target may not be installed
  = help: consider downloading the target with `rustup target add x86_64-unknown-uefi`

and

rustup target add x86_64-unknown-uefi
info: component 'rust-std' for target 'x86_64-unknown-uefi' is up to date

Any hint?

closed 2026-09-29 by crawfxrd