system76/firmware-open #633: Firmware fails to update on System76 DarterPro (darp10-b)

cement-head, 2025-07-23:
- Model:  darp10-b
- BIOS version:  cat /sys/class/dmi/id/bios_version
- EC version: 
- OS:  Pop!_OS 22.04.2 LTS
- Kernel:  Linux 6.12.10-76061203-generic #202412060638~1751390742~22.04~79b9668 SMP PREEMPT_DYNAMIC Tue J x86_64 x86_64 x86_64 GNU/Linux

**### Firmware Update (2025-05-30_d5e14a9) on darp10-b fails**

### Steps to reproduce

Got a notification from my Firmware GUI that a firmware update was/is available and a green UPDATE button. After attempting to UPDATE I get this error message: **error in system76-firmware client: calling Schedule method failed: failed to add boot entry: exit status: 5**

[image]

### Expected behavior

Firmware should be updated on Reboot

### Actual behavior

Error on attempted reboot.

### Additional info

I think the solution is outlined here: https://support.system76.com/articles/open-firmware-smmstore/

But I will need a PATCHED firmware updater for darp10-b

cement-head, 2025-07-24:
$ efibootmgr
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0006,0005,0004,0000,0003,0001
Boot0000* Linux Boot Manager
Boot0001* UiApp
Boot0002* CT500P3PSSD8 
Boot0003* Samsung SSD 990 PRO 4TB 
Boot0004* Pop!_OS 22.04 LTS
Boot0005* Pop!_OS 22.04 LTS
Boot0006* Pop!_OS 22.04 LTS

cement-head, 2025-07-24:
I had to rescue/refresh my OS three times as I was trying to install the Intel oneAPI (but it completely broke my system - FYI: Don't try installing it).  Anyway, after that I had to manually reorder the boot order.  Does that help? 

Sorry, I know absolutely nothing about EFI and how computers boot.

crawfxrd, 2025-07-24:
Yeah those `dump-type0` entries have filled the 64 KiB working block for SMMSTORE. (I thought we had 128 KiB usable space, but apparently not.)

```
df -h /sys/firmware/efi/efivars/
```

If you don't need the dumps, you can try deleting them with:

```
sudo rm /sys/firmware/efi/efivars/dump-type0*
```

cement-head, 2025-07-24:
```
$ df -h /sys/firmware/efi/efivars/
Filesystem      Size  Used Avail Use% Mounted on
efivarfs         64K   59K    32 100% /sys/firmware/efi/efivars

```
After deleting the dumps

```
$ df -h /sys/firmware/efi/efivars/
Filesystem      Size  Used Avail Use% Mounted on
efivarfs         64K   59K    32 100% /sys/firmware/efi/efivars

```
```
$ ls -lSr /sys/firmware/efi/efivars/
total 0
-rw-r--r-- 1 root root     5 Jul 18 11:06 VendorKeys-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     5 Jul 18 11:06 VarErrorFlag-04b37fe8-f6ae-480b-bdd5-37d98c5e89aa
-rw-r--r-- 1 root root     5 Jul 18 11:06 SetupMode-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     5 Jul 18 11:06 SecureBoot-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     6 Jul 18 11:06 Timeout-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     6 Jul 18 11:06 BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     7 Jul 18 11:06 PlatformLang-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     8 Jul 18 11:06 MTC-eb704011-1402-11d3-8e77-00a0c969723b
-rw-r--r-- 1 root root     8 Jul 18 11:06 Lang-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     8 Jul 18 11:06 certdbv-d9bee56e-75dc-49d9-b4d7-b534210f637a
-rw-r--r-- 1 root root     8 Jul 18 11:06 certdb-d9bee56e-75dc-49d9-b4d7-b534210f637a
-rw-r--r-- 1 root root     8 Jul 18 11:06 BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    12 Jul 18 11:06 OsIndicationsSupported-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    12 Jul 18 11:06 OsIndications-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    12 Jul 18 11:06 LoaderFeatures-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    17 Jul 18 11:06 LangCodes-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    18 Jul 18 11:06 Key0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    18 Jul 18 11:06 BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    20 Jul 18 11:06 LoaderTimeInitUSec-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    20 Jul 18 11:06 LoaderTimeExecUSec-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    22 Jul 18 11:06 PlatformLangCodes-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    24 Jul 18 11:06 LoaderFirmwareType-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    28 Jul 18 11:06 LoaderFirmwareInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    34 Jul 18 11:06 ErrOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    34 Jul 18 11:06 ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    34 Jul 18 11:06 ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    44 Jul 18 11:06 LoaderEntrySelected-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    48 Jul 18 11:06 LoaderImageIdentifier-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    53 Jul 18 11:06 ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    66 Jul 18 11:06 ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    68 Jul 18 11:06 SignatureSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    70 Jul 18 11:06 Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    78 Jul 18 11:06 LoaderDevicePartUUID-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root    98 Jul 18 11:06 Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   112 Jul 18 11:06 PlatformRecovery0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   120 Jul 18 11:06 Boot0003-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   126 Jul 18 11:06 LoaderInfo-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root   152 Jul 18 11:06 LoaderEntries-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root   162 Jul 18 11:06 Boot0006-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   162 Jul 18 11:06 Boot0005-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   162 Jul 18 11:06 Boot0004-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   164 Jul 18 11:06 Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw------- 1 root root   516 Jul 18 11:06 LoaderSystemToken-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw------- 1 root root   516 Jul 18 11:06 LoaderRandomSeed-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
-rw-r--r-- 1 root root   923 Jul 18 11:06 PKDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  2491 Jul 18 11:06 KEKDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  4066 Jul 18 11:06 dbDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 21174 Jul 18 11:06 dbxDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
```

crawfxrd, 2025-07-24:
I don't know how pstore works.

Reboot and check if it's still full?

cement-head, 2025-07-24:
FIRMWARE Updated as expected behaviour.  Thanks for all your help.  I will also ask to close out the System76 Support Ticket and reference this issue for their records.

[image]

cement-head, 2025-09-29:
Getting a prompt every reboot that there is a "Update Available" - I believe this is related to this last problematic firmware update (see above).  How would I go about "clearing" this message?